openhwfoundation/cv32e40p #259: Gated clocks as generated clocks

HazemFathy, 2020-03-15:
Dears,
In cluster_clock_gating wrapper , i manually instantiated custom clock gating cells from STD lib. do i have to define the outputs of each cell as a generated clock with latency value for it? or i'll just leave it unconstrained and synthesis tool will identify it.

davideschiavone, 2020-03-16:
Hello,
I have never dono it, it should not be necessary to do so. 
I usually create a clock assigned to clk_i and then I synthesize the design. 
It would be nice if you can tell use whether you can see differences in defining the clock as output port of the clock gate cell.

Best 
Davide

HazemFathy, 2020-03-22:
I found that its not required to do that, however, large skew appears during CTS phase. it appears that ideal latency is not balance for both setup / hold violations.

davideschiavone, 2020-03-22:
Thanks for sharing this with us. Do you think it makes sense to create a clock as output of the clock gating cell? does it help with balance setup/hold violations?
Can you also tell us the tool you used please?

HazemFathy, 2020-03-22:
Thanks Davide,

I found some people that say it is known to define the output of clock gating cells as a generated clocks while others are saying that generated clocks are only for clock dividers , multipliers..etc. that generate a different clock domain. but defining them as generated clocks didn't adjust the latency for gated and ungated paths, i'm working on IC compiler with clock gating flow ( synthesis using compile_ultra -gate_clock) 

Problem is that ideal latency is not adjusted properly for hold checks and setup checks , because of the large global skew between ungated paths and gated paths.

Thanks,
Hazem

HazemFathy, 2020-03-23:
Thanks all,

I performed a few skew optimizations and it passed correctly.

davideschiavone, 2020-05-17:
@HazemFathy can we close the issue now?

HazemFathy, 2020-05-17:
Yes , thanks davide

closed 2020-05-17 by HazemFathy