openhwfoundation/cv32e40p #270: Testbench Makefiles should us the common manifest (cv32e40p_manifest.flist)

MikeOpenHWGroup, 2020-03-19:
This issue is related to the discussion for PR #266.

It would good to update the Makefiles below the `tb` directory to use the `cv32e40p_manifest.flist` manifest.   That will ensure Designers and Verifiers are compiling the same code.  For example, this is (similar to) what the simulation Makefiles in [core-v-verif](https://github.com/openhwgroup/core-v-verif) have:
```
# RTL source files for the CV32E core
# DESIGN_RTL_DIR is used by CV32E40P_MANIFEST file
PROJ_ROOT_DIR = $(PWD)/../../..
CV32E40P_PKG         := $(PROJ_ROOT_DIR)/core-v-cores/cv32e40p
FPNEW_PKG            := $(CV32E40P_PKG)/rtl/fpnew
CV32E40P_MANIFEST    := $(CV32E40P_PKG)/cv32e40p_manifest.flist
export DESIGN_RTL_DIR = $(CV32E40P_PKG)/rtl

# Verilator compile target
testbench_verilator: $(CV32E40P_PKG) $(TBSRC_VERI) $(TBSRC_PKG)
	$(VERILATOR) --cc --sv --exe \
		$(VERI_TRACE) \
		--Wno-lint --Wno-UNOPTFLAT \
		--Wno-MODDUP --top-module \
		tb_top_verilator $(TBSRC_VERI) \
		-f $(CV32E40P_MANIFEST) \
		$(TBSRC_CORE)/tb_top_verilator.cpp --Mdir $(VERI_OBJ_DIR) \
		-CFLAGS "-std=gnu++11 $(VERI_CFLAGS)" \
		$(VERI_COMPILE_FLAGS)
	$(MAKE) -C $(VERI_OBJ_DIR) -f Vtb_top_verilator.mk
	cp $(VERI_OBJ_DIR)/Vtb_top_verilator testbench_verilator
```

Silabs-ArjanB, 2020-06-16:
Note: This discussion is continued in https://mattermost.openhwgroup.org/all-users/channels/twg--cores--manifest

MikeOpenHWGroup, 2020-11-03:
The `core` testbench and `uvmt_cv32` verification environment in [core-v-verif](https://github.com/openhwgroup/core-v-verif) both use the **_cv32e40p_manifest.flist_** to compile the core, but the Makefilel in the example_tb does not.

MikeOpenHWGroup, 2020-11-17:
Hi @Silabs-ArjanB, @davideschiavone, I have removed the `Type:Enhancement` label as I am unconfortable waiving this issue for RTL freeze.  There is a chance, however slight, that users of the example_tb in this repository will be compiling/simulating a different version of the cv32e40p than does the uvm testbench in core-v-verif.

The last comment on the MatterMost channel discussion topic indicated that a resolution of this issue was a priority and I certainly agree with that.

Silabs-ArjanB, 2020-11-17:
Hi @MikeOpenHWGroup Where can I find documentation on the format used in ".flist"? The format/methodology agreed upon was FuseSoc (which does not imply .flist as far as I know, but please correct me if I am wrong) and unfortunately support for that did not materialize. Doing FuseSoc related changes is out of scope for RTL freeze as far as I am concerned.

MikeOpenHWGroup, 2020-11-17:
The "flist" format is a quasi-standard file format supported by all SV simulators (although I do not know if that includes Verilator).  So, for example, `xrun -f manifest_file` compiles all source files listed in the file.  This is essentially what we should see in the Makefile.

I am in favor of adopting FuseSoC for OpenHW projects, but that did not happen for cv32e40p (yet).  It is under consideration for future versions of core-v-verif.   In any case, the FuseSoC discussion is not relevant to this particular issue.

davideschiavone, 2021-03-03:
PR #630  should solve this