openhwfoundation/cv32e40p #280: Remove parameters and code for the old FPU

davideschiavone, 2020-03-30:
The core used to be interfaced with an FPU no longer supported.

https://github.com/pulp-platform/fpu/commits/master

in favor of fpnew , developed and maintained by @stmach , available at 

https://github.com/pulp-platform/fpnew

The core has still parameters and code that have been designed for the older FPU, thus they should be cleaned up.

Silabs-ArjanB, 2020-09-30:
Part of this work has been done in https://github.com/openhwgroup/cv32e40p/pull/528

I am aware that cv32e40p_alu.sv and include/cv32e40p_pkg.sv still need to be cleaned up as well (maybe there are other places as well that I am not aware about).

davideschiavone, 2020-10-02:
Second part need to be merged #536 before closing this issue

davideschiavone, 2020-10-28:
#536  has been merged. Closing this issue

closed 2020-10-28 by davideschiavone