openhwfoundation/cv32e40p #289: Limit CPU to not prefetch over 1KB boundaries?

Silabs-ArjanB, 2020-04-01:
Currently the cv32e40p has a prefetcher that will always try to keep its prefetch buffer as full as possible, thereby potentially prefetching over certain address boundaries. 

In a typical system slave modules are aligned to 1KB (or larger, e.g. 4KB, 16KB) boundaries. Suppose the cv32e40p is executing instructions from a RAM and a prefetch address goes over the actual RAM size into another peripheral (or undefined address space); this could lead to unwanted read-side effects or unwanted bus error (or interrupt) generation because of an access to the undefined address space. 

Limiting the instruction prefetch unit to never cross a (for example) 1KB address boundary would enable us to prevent such issues.

jm4rtin, 2020-04-01:
Wouldn't it be better to leave a gap in your memory map for this purpose? Altering the performance of the core at some arbitrary boundary sounds less than ideal. It would impact performance of any code at those locations as well as complicate the logic in the prefetcher.

Silabs-ArjanB, 2020-04-01:
We would indeed normally have a gap in the memory map, but a regular bus interconnect will give you a bus error if you perform an access to such a gap. (As cv32e40p does not have a bus error input pin you could convert such a bus error into an interrupt, and the cv32e40p will have no way of knowing of whether the interrupt was caused by a 'speculated' fetch or an actual wrong fetch).

bluewww, 2020-04-01:
I think it makes sense to document this potential issue and forumulate some requirement such as "there needs to be a x byte size gap between executable code and peripheral address spaces" because the prefetcher can touch the next x bytes and cause side effects (depending on the read semantics of the peripheral reg that might be touched).

Also I think we need to add a bus error signal anyway, but this is another topic.

jm4rtin, 2020-04-01:
Another option would be to restrict access to the peripheral space so that the instruction bus has no access to the peripheral registers in the first place. This situation would normally flag a bus error - a feature I would also like to see.

davideschiavone, 2020-04-01:
I also think it's better to add a bus error signal to the core rather than hacking the prefetcher as a work around to solve another source of problems.
The PMP for instance does something similar, if the prefetcher fetch an area that is not allowed to fetch, it stops but an error is triggered only if that instruction is actually executed by the core (for instance, if an instruction after a taken branch is not meant to be executed, of course it's fetched, but it triggers an error only if it goes inside the ID stage).

bluewww, 2020-04-01:
@davideschiavone care to open issue to track the progress of adding a bus error signal?

davideschiavone, 2020-04-01:
@Silabs-ArjanB , is it a feature for the cv32e40p or not?
In my opinion, it can be done as a long term plan

Silabs-ArjanB, 2020-04-02:
I was also thinking about requesting a bus error on the interface (again) indeed. We would definitely benefit from it on the instruction interface side, but for orthogonality we should probably also add it on the data interface side.

My preference would actually be to re-add bus error pins (exactly as they used to exist on RI5CY before the PMP was introduced) and that we (for now) solve this topic as a 'warning for the user in the documentation'.

davideschiavone, 2020-04-02:
Yes let's add this feature. It's cleaner. We can think about the bus error as an extra bit (33th bit) of the RDATA.

MikeOpenHWGroup, 2020-04-02:
@Silabs-ArjanB , @davideschiavone @bluewww @jm4rtin, I also like this idea.  _Please_ start by documenting it and updating the user_guide as appropriate _before_ updating the RTL.

Our documentation is already very thin and incomplete and this is a significant risk for the project.

Silabs-ArjanB, 2020-05-19:
Conclusion for this ticket:

- Will not limit prefetch behavior related to 1KB boundaries
- Will document prefetch behavior and warn about the issue raised here(done in https://github.com/openhwgroup/core-v-docs/pull/37)
- Will issue a feature request to implement bus errors on future cores (see #343)

closed 2020-05-19 by Silabs-ArjanB