openhwfoundation/cv32e40p #300: Compilation fails for rvc.S

Silabs-ArjanB, 2020-04-08:
arbink@etxaus037:[core](ArjanB_parameter_cleanup)> /tool/pulp_riscv_gnu_toolchain/2019.12/rhel6-64/bin/riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (GCC) 7.1.1 20170509

The following leads to an internal compiler error in my setup (in today's master repos):

cd cv32e40p/tb/core
make firmware-vsim-run 

....

/tool/pulp_riscv_gnu_toolchain/2019.12/rhel6-64/bin/riscv32-unknown-elf-gcc -c -march=rv32imc -g -o riscv_tests/rv32uc/rvc.o
-Iriscv_tests/ -Iriscv_tests/macros/scalar
-DTEST_FUNC_NAME=rvc
-DTEST_FUNC_TXT='"rvc"'
-DTEST_FUNC_RET=rvc_ret riscv_tests/rv32uc/rvc.S
riscv32-unknown-elf-gcc: internal compiler error: Floating point exception (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See https://gcc.gnu.org/bugs/ for instructions.
make: *** [riscv_tests/rv32uc/rvc.o] Error 4

I should have a recent PULP gcc install, although I wonder if that is really true given the following (could you please comment on that?):

/tool/pulp_riscv_gnu_toolchain/2019.12/rhel6-64/bin/riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (GCC) 7.1.1 20170509

I am using rvc.S as on https://github.com/riscv/riscv-tests. Removing the  .option norelax seems to help.

MikeOpenHWGroup, 2020-04-08:
Is there any value in pursing this?  This core is being verified under the [core-v-verif](https://github.com/openhwgroup/core-v-verif) repo and in that repo there are two project, [CV32E40P XPULP ISA Compliance](https://github.com/openhwgroup/core-v-verif/projects/5) and [CV32E40P RISC-V ISA Compliance](https://github.com/openhwgroup/core-v-verif/projects/4) which specifically target issues like this one.  These project are stalled by the lock-down in France, but we can be hopeful that our colleagues will be able to return to their offices before too long.

What we agreed to for this repo was a simple "hello_world" style "example testbench", which goes not require this issue to be fixed.

davideschiavone, 2020-04-08:
I agree with @MikeOpenHWGroup , the modifications to the riscv_test suite should be done in the core-verif. Only hello world should work here.

Silabs-ArjanB, 2020-04-09:
Also agreed with that. Does this test work in the core-v-verif repos. If so, then this issue can be closed; if not, the we can re-open it in the correct repos.

Silabs-ArjanB, 2020-06-15:
Issue will be closed (might show up again in core-v-verif if not automatically solved in the new compliance test setup once that is available)

closed 2020-06-15 by Silabs-ArjanB