openhwfoundation/cv32e40p #302: Retire signal not correct for misaligned accesses

GTumbush, 2020-04-08:
The attached code does a misaligned access to memory.  The address accessed is 0x47F31. According to the RI5CY User Manual:

*The LSU is able to perform misaligned accesses, meaning accesses that are not aligned on natural word boundaries. However, it needs to perform two separate word-aligned accesses internally. This means that at least two cycles are needed for misaligned loads and stores.*

However, riscv_tracer triggers the Retire event after the first word-aligned access. It should not trigger the Retire event until after the second word-aligned access.  In addition a nonsense value is written to the insn_regs_write queue.

Referring to the waveform below:
1. Prior to the lw x15,-40(x8) instruction being executed relevant GPR values are:
   1. x8= 0x47F80
   1. x15=0
1. Instruction lw x15,-40(x8) is flagged as Retired at time 1,683,552ns
   1. addr=0xF and value=0x47f31 is written to the insn_regs_write queue.
   1. This is correct.
1.  Instruction lw x15, 0(x15) is flagged as Retired at time 1,683,650ns. However,
    1. The 2nd word-aligned access has not completed
    1. A nonsense value of 0x47F has been written to the insn_regs_write queue
    1.  The expected value written to the insn_regs_write queue is 0x04030201
1. Instruction addi x11, x15, 0 is flagged as Retired at time 1,683,811ns
    1. addr=0xB and value=0x04030201 is written to the insn_regs_write queue.
    1. This is correct

[image]

eroom1966, 2020-04-09:
@davideschiavone 
Hi Davide, could you please provide some feedback on this issue that Greg and myself have been investigating
Thx
Lee

davideschiavone, 2020-04-14:
Working on it! Let you know asap!

davideschiavone, 2020-04-20:
Hi there, 
I simulated this scenario and indeed it was wrong, good catch thanks!
Here in my repository I have a fix

https://github.com/davideschiavone/riscv/commits/fix%23302

May you please try it? This branch contains also the fix about the csr updates mentioned here #256 
as this is not yet merged in the master.

Let me know please, 
Davide

GTumbush, 2020-04-22:
Hi Davide,
  This branch does fix the misalign issue. 

 Does this branch contain the fix regarding the fence instruction #255 ?

Thanks,
Greg

davideschiavone, 2020-04-22:
Hi,

yes!

It's probably safer if we merge these two fixes and then fix a posteriori in case the fix for #256 introduced bugs. Otherwise we risk to lose them.

What do you think @Silabs-ArjanB  and @MikeOpenHWGroup ?

MikeOpenHWGroup, 2020-04-22:
@davideschiavone, I agree that this is a reasonable path forward.   Keep in mind that this won't automatically break verification because I need to manually move the git hash used in our Makefiles to point to a specific version of cv32e40p.   Once this merge happens, I'll update and report back.

GTumbush, 2020-04-22:
Hi Davide,
Please confirm that fix#302 contains fix for #255.  I ran our testcase and do not see the fence instruction being logged in the tracer output.  Below is the sequence I used to get fix #302

git clone https://github.com/openhwgroup/cv32e40p.git
cd cv32e40p
git remote add davideschiavone https://github.com/davideschiavone/riscv.git
git fetch davideschiavone
git merge davideschiavone/fix\#302
make deps

Is this correct? 

Adding @eroom1966

davideschiavone, 2020-04-22:
You are right, somehow it was not been merged (my mistake).

Now it's part of the master. I merge this PR as well so that you can use the latest master in the openhw repository that contains all the fixes.

As @MikeOpenHWGroup said, this won't break verification. In addition, it's a corner case that should anyway be tested in a special way.

Best
Davide

GTumbush, 2020-04-22:
OK, now if I merge fix302 with master the misalign (#302), Retire ( #256 ) and fence ( #255 ) are fixed.

davideschiavone, 2020-04-26:
Fix and merged now: https://github.com/openhwgroup/cv32e40p/pull/313

closed 2020-04-26 by davideschiavone