openhwfoundation/cv32e40p #309: non-compliant MSTATUS UIE/UPIE, MISA[N]=0 ?

eroom1966, 2020-04-20:
Hi 

It appears the the MISA[N] bit is hardwired to zero, but it is still possible to write the UIE and UPIE bits in the MSTATUS register

We are running the Compare testbench between riscvOVPsim and cv32e40p RTL, and we observe a difference here

> Info 'root/cpu0', 0x00000000000168cc(init_machine_mode+4): 8ba20213 addi    tp,tp,-1862
> Info   tp 00102000 -> 001018ba
> Info 'root/cpu0', 0x00000000000168d0(init_machine_mode+8): 30021073 csrw    mstatus,tp
> Info   mstatus 00001800 -> 00001888
> Info 'root/cpu0', 0x00000000000168d4(init_machine_mode+c): 00001237 lui     tp,0x1
> **7750: ERROR: mstatus expected=0x00001888 and actual=0x00001898 PC=0x000168d0**
> Info   tp 001018ba -> 00001000

This is telling us that the MSTATUS UPIE bit is set to '1' in the RTL, but set to '0' in the reference. The ISA spec says the following 

> User-level interrupts are an optional extension and have been allocated the ISA extension letter N. If user-level interrupts are omitted, the **UIE and UPIE bits are hardwired to zero**.

So it seems these conditions are mutually incompatible, without MISA[N], UIE and UPIE are hardwired to zero

I note a related issue was raised back in Sept/19
https://github.com/openhwgroup/cv32e40p/issues/140
Thx
Lee

davideschiavone, 2020-04-20:
Hello,
the core does not support the UMode, indeed we updated the core to prevent the users of the core to set PULP_SECURE to 1 as it may generate confusion as indeed happened here.

With PULP_SECURE hardwired to 0 as it is today it should not be possible to write to U registers.

Are you using the last version of the core?

Davide

Imperas, 2020-05-01:
This is resolved, I see this is now hardwired to 0
please go ahead and close this issue

closed 2020-05-01 by eroom1966