openhwfoundation/cv32e40p #314: doc: user_manual

eroom1966, 2020-04-23:
A few comments on the user manual, I note the redline doc was committed, was this intentional ?
I see some issues in this doc, as follows

for cv32e40p PULP_SECURE - (as I understand it) PULP_SECURE is now hardwired to 0, in which case should all references to PULP_SECURE be removed, including any dependencies, such as all references to USER mode state

section 10.2 lists register USTATUS, this is not listed in the full CSR listing 10. and should no longer be present as USER mode is not present?

the above is also true for UTVEC, UEPC, UCAUSE

Thx
Lee

MikeOpenHWGroup, 2020-04-23:
Hi Lee, thanks for this.  We treat problems in the documentation just as seriously as if it was in the RTL, so creating this issue was the right thing to do.    A few points for future issues:
- Please assign all documentation issues to @davideschiavone.  He will re-assign as needed.
- Please also add a "Documentation" label.
- Lastly, and this is the most important, if there are multiple problems with the docs, create a separate issue for each one.

davideschiavone, 2020-04-24:
Good point indeed, your question makes completly sense @eroom1966 and thanks @MikeOpenHWGroup for adding clarifications.

I would say "yes, let's remove them", but the idea of leaving the RTL there (instead of cleaning it up) was to let other uses to fork the core and have a starting point for what concerns the hardwired parameters. The same would be true for the documentation as well. 
I propose to mark them as "not available by default" in the documentation or something like this.
What is your opinions? @MikeOpenHWGroup @eroom1966 @Silabs-ArjanB

Silabs-ArjanB, 2020-04-24:
I would be fine with removing the content (that would be clearest in my opinion). I don't know if our current documentation format would allow for 'conditional inclusion'; if that is possible, then we would at least have a way to add the content back in easily for a next CV32 core.

MikeOpenHWGroup, 2020-04-24:
> [does] our current documentation format would allow for 'conditional inclusion'

I have not try this, but the following should work:
```
My Chapter
=========

This is a chapter, bla bla bla

.. this is conditional inclusion text
   it stays in the .rst source, but is not rendered to html, pdf, etc. by Sphinx and RTD

This is more stuff in my chapter that is rendered...
```

Silabs-ArjanB, 2020-04-26:
Then using conditional inclusion would be my preference for content that is no longer part of CV32E40P (e.g. user mode, PMP, etc.). If this is considered too much work I am fine with completely removing such no longer content all together. I think we should not render out-of-date content into the html/pdf, even not if we add some kind of warning that specific content does not actually apply to CV32E40P.

Silabs-ArjanB, 2020-05-07:
I found that we can use conditional exlcusion, e.g we could insert the following before the parts referring to user mode:

.. only:: USER

The user mode documentation will then only get rendered when sphinx-build is called with '-t USER'.

I added a specific User mode related topic issue at https://github.com/openhwgroup/core-v-docs/issues/39, so I believe we can close the ticket here (to avoid duplicates).

MikeOpenHWGroup, 2020-05-07:
> openhwgroup/core-v-docs#39

Slick!   Thanks for the tip Arjan.

---mike

davideschiavone, 2020-05-17:
Issue duplicated in https://github.com/openhwgroup/core-v-docs/issues/39 and thus closed

closed 2020-05-17 by davideschiavone