openhwfoundation/cv32e40p #317: Dead states in load/store unit

Silabs-ArjanB, 2020-04-27:
I am studying the current load/store unit in preparation for OBI related fixes. I see (using formal verification) that the WAIT_RVALID_EX_STALL and IDLE_EX_STALL can never be reached.

Maybe they were reachable at some point (e.g. before we disallowed certain parameter settings as e.g. USE_PMP). Can somebody explain the (original) reason for these states (before I will proceed with removing them)?

davideschiavone, 2020-04-27:
Hello @Silabs-ArjanB ,

there are other people (you included btw) that also found this and reported it here:

https://github.com/openhwgroup/cv32e40p/issues/129
https://github.com/openhwgroup/cv32e40p/issues/80

We tried to reach with random instruction simulation that state back in 2017 and failed indeed.
We did not remove it as we did not investigate further with formal verification as well (or synthesis could work too as it should remove those states)

The designer left so we do not know the original reason . Probably it comes from the very first version based on OpenRISC. Maybe @gautschimi knows it?

I think we should remove them once we verify that Synthesis or Formal Tools confim that.

Best
Davide

Silabs-ArjanB, 2020-04-28:
Hi @davideschiavone , I knew I reported something similar quite some time ago, didn't realize it was exactly the same, sorry for that. I also understand the reasoning for the lsu_ready_ex_o I asked about now (I will remove that from the original post). This ticket can be closed then.

closed 2020-04-28 by Silabs-ArjanB