openhwfoundation/cv32e40p #319: Disable custom extensions

Imperas, 2020-04-29:
If I understand correctly it is possible to optionally **include** the extensions for F and A
is it also possible to **exclude** the "PULP specific extensions" ?

thx
Lee

Imperas, 2020-04-29:
@MikeOpenHWGroup 
Mike in an earlier thread, you asked me to assign a label to my issue.
I do not seem to have access to assign labels - do I need to be authorized ?

Silabs-ArjanB, 2020-04-30:
Hi, please see table 2 of https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/CV32E40P_and%20CV32E40_Features_Parameters.pdf . The idea is that the PULP ISA extensions are unconditionally included, except for 'hardware loop functionality', which will have a parameter to decide whether it is included or not  (which is about to be added to the RTL).

Imperas, 2020-04-30:
thanks for the pointer

closed 2020-04-30 by Imperas