openhwfoundation/cv32e40p #320: Non-Maskable Interrupt

Imperas, 2020-04-30:
Looking at the RTL I see a signal at the top level for irq_nmi_i but I do not see any description in the manual for NMI, in particular when asserted does the core react to this interrupt, and if so what is the vector address for NMI ?
thx
Lee

MikeOpenHWGroup, 2020-04-30:
Hi Lee.  When creating an issue, please assign it explictly to ensure someone picks it up.  When in doubt, assign it to either @davideschiavone or @MikeOpenHWGroup and we can re-direct as necessary.

You are right that the [user manual](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/exceptions_interrupts.html#exceptions-and-interrupts) does not yet capture the interrupts.  This is a known item on the ToDo list.   In the meantime, you can check out the equivalent functionality in [IBEX](https://ibex-core.readthedocs.io/en/latest/exception_interrupts.html?highlight=irq_nm_i) as I believe cv32e40p will be quite similar.  (@davideschiavone , please correct me if I wrong on this!).

davideschiavone, 2020-05-01:
Yes it is correct, let's move this issue in the core-docs as this is a missing features in the current documentation.

best
Davide