openhwfoundation/cv32e40p #321: Question regarding DM_HALTADDRESS

MikeOpenHWGroup, 2020-04-30:
I've received a [pull request on core-v-verif](https://github.com/openhwgroup/core-v-verif/pull/100) to change the value of the core's DM_HALTADDRESS parameter from 0x1A110800 to 0x000F0000.  The UVM testbench has a 1Mbyte address space, so this will set the debug page to the last 64K which seems very reasonable to me.

However 0x1A110800 is a very particular number - is there any reason it was used?   Are there implications to the DUT or TB if this gets changed?

davideschiavone, 2020-05-01:
This is a parameter of the core so it can be changed in whatever number from the instantiator module or testbench.

This number `0x1A110800 ` is where the debug module is mapped in PULP, so as the core was part of the pulp project we left as default parameter what we have in PULP.

bluewww, 2020-05-01:
Changing `DM_HALTADDRESS` in the core changes the debug exception address (the address the core jumps to when the debug request line is asserted). See
https://github.com/openhwgroup/cv32e40p/blob/8666456b147e4ed32590d10c920ea6c919761bc4/rtl/riscv_if_stage.sv#L145

Changing this value in the TB means making sure that the debug module is mapped at that location in the address space and also that the core is made aware of this new value.

MikeOpenHWGroup, 2020-05-01:
Thanks @bluewww.  A couple of questions:
1. Why was 0x1A110800 selected for the debug module halt address?    Are there restrictions on the debug module that require a specific address?
2. When you say `...and also that the core is made aware of this new value`, does this mean setting the value of the DM_HALTADDRESS parameter in the RTL or something else?

bluewww, 2020-05-01:
> Thanks @bluewww. A couple of questions:
> 
>    Why was 0x1A110800 selected for the debug module halt address? 

You mean why it was chosen as default address? No particular reason except that it is the address used in PULP. Default values don't matter that much anyway since your supposed to chose your own when instantiating the core. Maybe it would be clearer if it was changed to some less arbitrary looking value.

>    Are there restrictions on the debug module that require a specific address?

No there are no restrictions, but when choosing the zero address then the debug module only requires one scratch register instead of two.

>   When you say `...and also that the core is made aware of this new value`, does this mean setting the value of the DM_HALTADDRESS parameter in the RTL or something else?

Yes I mean setting `DM_HALTADDRESS`

Silabs-ArjanB, 2020-05-18:
FYI: Related to this there is a proposal in https://github.com/openhwgroup/core-v-docs/issues/52
to replace the DM_HALTADDRESS parameter by a dm_halt_addr_i pin.

The reasons for this proposed change are as follows:

    Treat the debug halt address in the same way as boot_addr_i (which is also a pin and not a a parameter)
    Increase coverage related to this functionality (it is easier to change the pins value than to change the parameter value for (formal) verification)

closed 2020-05-29 by davideschiavone