openhwfoundation/cv32e40p #323: NMI is masked by MSTATUS[MIE]

Imperas, 2020-05-01:
@davideschiavone @MikeOpenHWGroup 
I think there is a bug with the implementation of the NMI interrupt

The interrupt is only taken if the MSTATUS[MIE] is enabled, this is not correct as NMI is by definition 'non-maskable'
From the privilege spec

> Non-maskable interrupts (NMIs) are only used for hardware error conditions, and cause an immediate jump to an implementation-defined NMI vector running in M-mode **regardless of the state of a hart’s interrupt enable bits.**

Thx
Lee

eroom1966, 2020-05-01:
Additionally it appears that before the NMI is asserted, 
MSTATUS[MIE]=1, MSTATUS[MPIE]=0

after the NMI is taken 
MSTATUS[MIE]=0, **MSTATUS[MPIE]=0**

but I would expect MSTATUS[MPIE]=1

Thx
Lee

Silabs-ArjanB, 2020-05-14:
Issue might relate to #336  as well.

Silabs-ArjanB, 2020-06-09:
Hi @Imperas The NMI functionality has been removed all together. Could you please close this issue?

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB