openhwfoundation/cv32e40p #325: MPEC and MSTATUS update before Retire signal for ebreak

GTumbush, 2020-05-01:
The tracer triggers Retire event correctly for ebreak instruction. But the MPEC and MSTATUS registers update on the Retire event before the ebreak instruction.  This causes a miscompare with the ISS.

[image]

The software to produce the above is in attached zip file.

[riscv_ebreak_test_0.zip](https://github.com/openhwgroup/cv32e40p/files/4566226/riscv_ebreak_test_0.zip)

GTumbush, 2020-05-01:
I cannot assign this. Please assign to Davide.

MikeOpenHWGroup, 2020-05-01:
Done.

GTumbush, 2020-05-11:
Hi David, Were you able to reproduce this error?

davideschiavone, 2020-05-11:
Sorry for later reply.
I am trying to fix it but I don't understand the difference between this behaviour and the fix we have done for #256 .

The ebreak is behaving exactly like the ecall, so the MSTATUS get's updated in FLUSH_WB, which be wrong also for ecall, why isn't it? Can you help me figured it out? the 2 instructions behave the same

GTumbush, 2020-05-11:
Hi Davide,
  I added to #256 a waveform with your fixes so we can see the corrected behavior.

This is the behavior of ecall. mstatus and mepc update on the rising edge of Clk immediately before the retire event on the falling edge of Clk.
[image]

This is the behavior of ebreak: mstatus and mepc update for the previous instruction.
[image]

davideschiavone, 2020-05-11:
Hi,

I think the problem is related to the REMU, and in general, to multicycle instructions preecing system calls or instructions that flush the pipeline

Here we have the same test you have with ECALL instead of EBREAK and indeed the error is the same

[image]

Now if I put the ADD instead of remu I get the correct behaviour with the EBREAK as well

[image]

So now I try to investigate what the problem is with REMU followed by EBREAK/ECALL

davideschiavone, 2020-05-11:
I have a very sentitive fix, but again, to fix the tracer I had to change the controller.
For this reason I need extra eyes looking at this change. 
May I ask to @Silabs-ArjanB, @MikeOpenHWGroup  and @bluewww  to have a look at this?
https://github.com/davideschiavone/riscv/commit/2661d8464ddf5af6bf438f346b05d38132f2a02b

Any implication? Maybe Robert can run his debug tests on it

Silabs-ArjanB, 2020-05-12:
@davideschiavone It is very difficult to review this without understanding at a high level when (i.e. in what pipeline stage) CSRs are supposed to be updated inside CV32E40P. For example the strategy could be that every instruction updates the CSR takes 2 cycles in the EXECUTE stage and performs the CSR update in the first EXECUTE cycle and retire in the second EXECUTE cycle. Then any instruction in ID will read the CSR correctly. Functionality that use a CSR in the IF stage (e.g. because they rely on PMP settings in a future chip) would require a pipeline flush. Could you please explain the high-level idea for dealing with CSRs and how your fix relates to that?

GTumbush, 2020-05-12:
Hi Davide, this fix does address the issue but creates many more later in the simulation.  Soon after the earlier error the PC stalls at an ebreak instruction. The PC no longer advances.

PCr_lower20 is the PC of the ISS.  insn_pc is from the RTL. 

[image]

davideschiavone, 2020-05-13:
Hi @GTumbush , 

unfortunalty I did modify the tracer sorry. As now these instructions depends on id_ready, I had to remove the OR condition in the `log execution` process.

I pushed another commit here:

https://github.com/davideschiavone/riscv/commit/0db91db7d28326474dc5e20c694b2dbdaeb9d45e

Can you please try? I know fixing and testing in two different set up takes a little bit longer time and extra effort but hopefully it's only at the beginning while creating this verification set up :)

best
Davide

davideschiavone, 2020-05-13:
> @davideschiavone It is very difficult to review this without understanding at a high level when (i.e. in what pipeline stage) CSRs are supposed to be updated inside CV32E40P. For example the [...]

Hi Arjan, 

I think that for clarity reasons we can penalize CSRs operations and make them all flush the pipeline so that they are all consistent as you suggest.

So far operations to CSRs can happen directly (exclicit writes to CSRs) or indirectly (instructions like ecall or interrupts etc).

When CSRs modify the MSTATUS or MEPC or DEBUG, the controller first goes to the FLUSH_EX state and waits for instructions in the EX to finish (e.g. loads waiting for grant, divisions etc). Then to FLUSH_WB where they actually makes things happen.

Now I proposed to stay in the DECODE until the ID stage is ready to prooced (whereas this was not the case before). As now the change to FLUSH_EX depends on id_ready.
This is now consistent with the other instructions as they update the ID/EX pipeline on id_valid (and so id_ready).

In FLUSH_EX now we tell the registers to update (so consisten with the other instructions) via the csr_save_id_o  signal. And in the next state FLUSH_WB, we actually do nothing for CSRs whereas we jump in case of traps.

To align all the CSRs operations to this behaviour we need to update the controller to add an extra cycle (as normally the would let the next instructions to proceed). Whereas now in FLUSH_EX and FLUSH_WB there won't be any valid instruction in the ID (so adding 2 bubbles).
It's a very high level explaination, but if you are interested in more details we can have a quick calls where we can share the screen and dig into details.

GTumbush, 2020-05-13:
Hi David, 
 Hash 0db91db fixes the issue with ebreak but it not correct for an illegal instruction. A snip of what is executed is below.
    89d0:       00812623                sw      s0,12(sp)
    89d4:       01010413                addi    s0,sp,16
    89d8:       0000                    unimp
    89da:       0000                    unimp
    89dc:       00000013                nop
    89e0:       00c12403                lw      s0,12(sp)

The behavior below is with Hash 0db91db
[image]

This is the correct behavior for the cv32e40p from the master branch.
[image]

The firmware is attached as illegal.zip 

[illegal.zip](https://github.com/openhwgroup/cv32e40p/files/4624646/illegal.zip)

davideschiavone, 2020-05-17:
Hello,

I think my output is correct, isn't it?

[image]

This is the trace

> 000001d2 00812623 sw               x8, 12(x2)
> 000001d6 01010413 addi             x8, x2, 16
> 000001da 0e0372b3 INVALID                    
> 00000000 3ee0006f   jal              x0, 1006  

Are you using this core here? https://github.com/davideschiavone/riscv/tree/fix325

GTumbush, 2020-05-17:
Hi Davide,
 The issue is that the INVALID instruction is marked as retired.  The PC should skip over the retired instruction. 

This trace is with fix325
316191  2636 000089d4 01010413 addi  x8, x2, 16 x8=00047f70  x2:00047f60
**316331  2637 000089d8 00010413 addi  x8, x2, 0   x8=xxxxxxxx  x2:00047f60**
316831  2641 00008100 4100006f   jal   x0, 1040           

This trace is with the master branch
316191 2636 000089d4 01010413 addi x8, x2, 16  x8=00047f70  x2:00047f60
316671 2641 00008100 4100006f jal x0, 1040           

Yes, I'm using the core at fix325

davideschiavone, 2020-05-17:
Ah I see what you mean. I misunderstood the issue.
So what do you prefer to have? A retired INVALID instruction or not as before?
Second question, can we keep in the tracer the INVALID instruction?

GTumbush, 2020-05-17:
Hi Davide,
 We want an illegal instruction to not be retired. This was the behavior before.
I'm OK with printing the INVALID instruction to the tracer file as long as the Retire event is not triggered.

 @eroom1966  please confirm

eroom1966, 2020-05-17:
Would an illegal instruction contribute to the minstret count ?
Doesn’t sound right, I would have thought it is trapped at the decode stage ?

GTumbush, 2020-05-20:
Hi Davide, what's the status of getting this fixed?
Thanks,
Greg

davideschiavone, 2020-05-29:
Hello, sorry for delay.
I pushed again in the fix325 branch to avoid retiring instructions that are illegal.

Can you please tell me if this helps?
thanks
Davide

GTumbush, 2020-05-29:
Hi Davide,
 I tried merging in fix325 with the cv32e40p repo master branch and got a merge conflict with riscv_core.sv. Here are the steps I took:

git clone https://github.com/openhwgroup/cv32e40p.git cv32e40p_fix325
cd cv32e40p_fix325
git remote add davideschiavone https://github.com/davideschiavone/riscv.git
git fetch davideschiavone
git merge 3b0ed25

This results in:
Auto-merging rtl/riscv_core.sv
CONFLICT (content): Merge conflict in rtl/riscv_core.sv
Automatic merge failed; fix conflicts and then commit the result.

Is this the correct steps?

Thanks,
Greg

davideschiavone, 2020-06-02:
This should now be merged, can you please try again?

GTumbush, 2020-06-03:
I'm assuming you mean merged with master. If so:

I ran the core-v-verif master testbench targeting the cv32e40p master. Below is the log of  cv32e40p master
commit f7aee5a7488bc3e58f4bfe5910b49ae850d02768
Merge: 9135b2f 9b21083
Author: Pasquale Davide Schiavone <pschiavo@iis.ee.ethz.ch>
Date:   Wed Jun 3 13:33:02 2020 +0200
    Merge pull request #356 from silabs-PaulZ/write_to_readonly_csr
    read only CSRs should return illegal instruction when written

The riscv_ebreak_test_0 test still fails as before. Questions:
1) Should I be targeting cv32e40p master?
2) Can you run the UVM testbench in core-v-verif comparing against the Imperas ISS?

Thanks,
Greg

davideschiavone, 2020-06-09:
Hi @GTumbush , somehow I didn't get the notification of this.
I couldn't run the core-v-verif yet, but I meant that the branch fix325 of the core on my repository has the master branch of openhw merged in.

Not the other way, as we merge only if it solves problem.
You should try this core: https://github.com/davideschiavone/riscv/tree/fix325

davideschiavone, 2020-06-11:
Hello @GTumbush , indeed. 
I am also running them.

By the way, I pushed again another fix and this solved also the illegal test.

I guess, if @MikeOpenHWGroup  agrees, you ( @GTumbush ) can still try if this core works and the I merge in master. 

Or if you guys think this is redundant I can simply merge into master, then create a PR to core-v-verif to use the last core.

best
Davide

davideschiavone, 2020-06-15:
Hi @MikeOpenHWGroup , @GTumbush , any comment on it?

commit 2661d846 (2020-06-15): :bug: :fire: fix #325, changes controller
closed 2020-06-15 by davideschiavone