openhwfoundation/cv32e40p #328: Fence.i interaction with external cache

haugoug, 2020-05-05:
In case the core is attached to an external cache, since OpenOCD just relies on fence.i when inserting breakpoints, is the core supposed to synchronize with the external cache when fence.i is executed by OpenOCD, and if so how ?

Silabs-ArjanB, 2020-06-16:
Hi @haugoug, the CV32E40P itself does not have cache(s) and FENCE.I therefore only flushes the internal core’s pipeline and prefetch buffer. The operation of FENCE.I cannot be observed on the CV32E40P boundary. The assumption is that external instruction memory is kept coherent in a user defined manner. In your scenario you could for example implement software-directed cache-flushing followed by a FENCE.I instruction.

On future cores we can consider adding a hardware-based handshake to aid in external flushing (I will add a feature request for a next core for that).

Could you please close this issue if you agree it has been answered in a satisfactory manner or otherwise let us know what information you are missing? (Without further feedback the issue will be closed automatically after 7 days.)

bluewww, 2020-06-16:
I think it is problematic that `fence.i` doesn't have a hardware interface, since now it is not possible to implement a system with a correctly working `fence.i`, when there is a cache between core and memory. Note that according to the RISC-V spec `fence.i` is supposed to do this synchronization between instruction and data stream transparently to the user.

Back when I implemented this instruction, I just made it work for PULPissimo, where there was only the prefetch buffer in the core that needed flushing and no I$. In the PULP cluster we have an I$ which we explicitely have to disable to have the correct `fence.i` semantics for the cores in there.

I kinda regret not adding a hardware interface back then and I recommend to do it at some point.

Silabs-ArjanB, 2020-06-16:
Hi @haugoug If the external cache has a flush mechanism that can be triggered by software (via a store instruction), then you can trigger the flush that way before executing the fence.i

haugoug, 2020-06-19:
Hi @Silabs-ArjanB. I agree with @bluewww that this is a problem that fence.i is not taking care of the external cache, as this is what is intended from a spec point of view. Due to that, openocd is just relying on fence.i, and we need to patch it to insert a SW cache flush and make it work with Pulp cluster.
What you propose in #366 is exactly what is needed, but I don't understand why you mention that CV32E40P will not have this feature, do you mean only other kind of cores will have it ?

Silabs-ArjanB, 2020-06-19:
My proposal is that future cores (e.g. CV32E40) will have this feature. The proposed change will imply changes to the controller increasing the required verification effort.

haugoug, 2020-06-19:
What is CV32E40 compared to CV32E40P ?

MikeOpenHWGroup, 2020-06-19:
Hi @haugoug.  Have a look at [CV32E40P and CV32E40 Features and Parameters](https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/CV32E40P_and%20CV32E40_Features_Parameters.pdf)

haugoug, 2020-06-29:
So from what I understand from this document, CV32E40P will be maintained but won't have any new feature, or CV32E40P will be dropped when CV32E40 is ready ? In particular, bug fixes will be done on both cores or just on CV32E40 ?

haugoug, 2020-06-29:
Ok thanks a lot for the information !

closed 2020-06-29 by haugoug