openhwfoundation/cv32e40p #329: tracer out of order for half-word aligned fence.i

eroom1966, 2020-05-08:
@davideschiavone @MikeOpenHWGroup @GTumbush 
Hi Davide,
It looks like we have found a similar issue to the bug found in #261 
In this example we have a fence.i instruction executing on a half-word aligned boundary
The code looks like this

```
3dbb6:       000a2783                lw      a5,0(s4)
3dbba:       00faa023                sw      a5,0(s5)
3dbbe:       0000100f                fence.i
```

When executing the tracer produces this
```
895357.000 ns 126675 0003dbb6 000a2783 lw       x15, 0(x20) x15=001101b3 x20:0008d760  PA:0008d760
895396.000 ns 126678 0003dbbe 0000100f fencei         PA:0003dbc6
895381.000 ns 126677 0003dbba 00faa023 sw       x15, 0(x21) x15:001101b3 x21:0003dbc2  PA:0003dbc2
```

as you can see the fence.i at 0x3dbbe is reported as executing before the sw at 0x3dbba
I suspect this is related either to the instruction ordering sequence of the lw/sw/fence - or the instruction alignment of the fence.i itself
thx
Lee

MikeOpenHWGroup, 2020-05-08:
Hi @eroom1966.  Good catch.  What test were you running?

eroom1966, 2020-05-08:
It was from @GTumbush running the compliance suite tests

GTumbush, 2020-05-08:
Test is cv32-riscv-compliance-tests

Waveform below shows PC 40f82->40f8a->40f86->40f8e 
[image]

davideschiavone, 2020-05-11:
Hello!

Thanks good catch! I'll try to fix it asap

davideschiavone, 2020-05-11:
Are you using the MASTER branch?

I tried to reproduce the same issue but I have no problem and the trace result is correct

GTumbush, 2020-05-11:
I just did a git pull origin in the sandbox and have the same erroneous behavior. I believe I'm up to date (see git.out attached).  Please paste your code and tracer output into this issue (as Lee has done).

[git.txt](https://github.com/openhwgroup/cv32e40p/files/4610879/git.txt)

davideschiavone, 2020-05-17:
Mh that's strange.

Can you do `git log` ?

Here is my trace

> 21216ns            2117 000001d2 000a2783 lw               x15, 0(x20)
21236ns            2119 000001d6 00faa023 sw               x15, 0(x21)
21246ns            2120 000001da 0000100f fencei

GTumbush, 2020-05-17:
Hi Davide,
 I was testing against the cv32e40p master with the following git log:
commit dde01442d11123763dccdf030b4bd3fbf4d9b4fe
Merge: 3952382 9d338f3
Author: Pasquale Davide Schiavone <pschiavo@iis.ee.ethz.ch>
Date:   Wed May 13 10:13:46 2020 +0200

    Merge pull request #331 from silabs-PaulZ/330_add_debug_trigger
    
    #330 add debug trigger

If I use https://github.com/davideschiavone/riscv/tree/fix325 I see the following which fixes this issue but now there are many miscompares before this instruction. With the cv32e40p master the fence issue was the first miscompare.

[image]

GTumbush, 2020-06-03:
Hi Davide,
  I ran the core-v-verif master testbench targeting the cv32e40p master. 
cv32e40p master
commit f7aee5a
Merge: 9135b2f 9b21083

The cv32-riscv-compliance-tests test still fails as before.

Can you run the UVM testbench in core-v-verif comparing against the Imperas ISS?

Thanks,
Greg

davideschiavone, 2020-09-21:
Hi @GTumbush , is that issue still present in the latest version of the core?

Silabs-ArjanB, 2020-09-24:
Hi @eroom1966 , @GTumbush Could you please let us know if this issue still present?

GTumbush, 2020-10-22:
Hello, due to network restrictions at EM I'm still not able to clone and run core-v-verif. If test cv32-riscv-compliance-tests passes this issue can be closed.

MikeOpenHWGroup, 2020-10-22:
I can confirm that the compliance regressions have been passing for some time.

Silabs-ArjanB, 2020-10-23:
> 
> 
> I can confirm that the compliance regressions have been passing for some time.

Unfortunately the compliance test is still failing as it has been for a very long time. I again made an issue for it in core-v-verif, https://github.com/openhwgroup/core-v-verif/issues/302

I am closing this specific issue here now as it then becomes a duplicate. (For what  it is worth; the test fails after the I_FENCE.I_01 part). I agree that the 'compliance test' is not such a great test, but it can be considered kind of a 'hello world'  type of test case and failing it sends a very poor message.

Silabs-ArjanB, 2020-10-23:
Closing this one and further tracking potential issue via https://github.com/openhwgroup/core-v-verif/issues/302

closed 2020-10-23 by Silabs-ArjanB