openhwfoundation/cv32e40p #334: Debug_req pulse width (bug)

silabs-PaulZ, 2020-05-12:
If the debug_req input signal is asserted for less than 5 clocks, the core does not properly enter debug mode. Depending on the pulse width, the core may fetch from the debug address (DM_HALT_ADDRESS) but may not the internal debug_mode bit.

The solution is to do one of the following:
1) Document a requirement that the debug_req must be asserted more than 5 clocks (which is reasonable assumption that debug_req is typically asserted for much longer, but may not cover all SOC implementations in the future and can lead to deadlock scenarios if not strictly adhered to)
2) Add a sticky debug_req capture register that is asserted until debug_mode is entered. It should then be cleared when debug_mode==1 and debug_req=0. This is a more robust solution to handle corner cases (especially shown in random verification)

Silabs-ArjanB, 2020-05-13:
I think '2' is the much nicer solution to this problem.

davideschiavone, 2020-05-13:
I also agree on the 2nd option. Interrupts are actually done in this way

PR #340 (2020-06-29): extend internal debug request until debug_mode is enterred
Thi PR fixes #334 

The debug_req_i is capture and extended until debug_mode is enterred.
closed 2020-06-29 by davideschiavone