openhwfoundation/cv32e40p #336: NMI bit should not be implemented in mie/mip register

abettati, 2020-05-13:
The [Interrupts_t](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/include/riscv_defines.sv#L445) structure is used to define [both mie and mip](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/riscv_cs_registers.sv#L287).
As it should be used only for hardware error conditions, I think it could be removed from both (as [IBEX](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_pkg.sv#L216) does). This also follows the specs as they label the upper bits in mie/mip as WPRI fields, meaning future standards can use them. 
One thing that I don't understand is why IBEX (from which I derived the RI5CY CLINT) [uses](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv#L476) them for fast lines: looking at the specs upper bits in  mie/mip should not be used.

Silabs-ArjanB, 2020-05-13:
In Ibex and CV32E40P the 16 upper bits of MIE/MIP are extended in a manner envisioned in the RISC-V specification. So this is a custom extension beyond CLINT (but in a manner recommended in the official spec).

abettati, 2020-05-13:
Hi @Silabs-ArjanB , thanks for the reply.
The only reference I found in the documentation I have, is in the [CLIC draft](https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc#clic-versus-original-basic-interrupt-controller):

> The existing original basic interrupt controller was a small unit that provided local interrupts based on earlier designs, and managed the software, timer, and external interrupt signals (xsip/xtip/xeip signals in the xip register). This basic controller also allowed additional custom fast interrupt signals to be added in bits 16 and up of the xip register.

Are you aware of a CLINT specific document from the foundation that I am missing? That would be nice to have:)

Silabs-ArjanB, 2020-05-13:
Hi @abettati I am basing my remarks above on the following text from the RISC-V privileged spec:

[image]

Best regards,
Arjan

abettati, 2020-05-13:
Ok, thanks.
For the NMI bit, I guess the same reasoning applies then.

Silabs-ArjanB, 2020-05-13:
Just added a note about NMI in my above post; that is indeed much less clear. We (and Ibex) are ahead of the official spec so likely we did something that will end up not being compatible to the official spec once it becomes available.

abettati, 2020-05-13:
Yeah thanks for pointing that out!
I was asking mainly for it, because I see that it is also named in a separate section in the specs (Sec. 3.4 Non-Maskable Interrupts). While I initially thought that handling it in the Interrupt_t struct was good to keep the code compact, I realized mie is probably wasting a bit (mie.irq_nmi is never used) and also that even mip wasn't strictly required to implement it.

abettati, 2020-05-13:
@Silabs-ArjanB I spot what I think is a small bug, again related to NMI, not going to open another issue for this.
NMI is globally masked in the [interrupt controller](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/riscv_int_controller.sv#L83) and in the core controller (for example [here](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/riscv_controller.sv#L446)m but actually every time the FSM checks for the interrupts to be enables).

The problem is that the [irq_pending](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/riscv_cs_registers.sv#L1161) signal inside the CSRs has no way to distinguish between NMI and regular lines. 
Probably NMI is not needed at the moment anyways, but in case I think this should be fixed. Just a heads up.

Silabs-ArjanB, 2020-05-14:
@abettati Agreed with your observation; that is a bug indeed. At first glance I think it might be a similar observation as the bug reported in #323 

Also agreed that we should align to Ibex here, although I wonder if we should skip nmi all together until it is officialy specified in the RISC-V organization; any opinions on that?

abettati, 2020-05-16:
@Silabs-ArjanB sorry for the delay, I missed the notification.
Non-maskable Interrupts are mentioned in Sec. 3.4 of the current specifications, but very little is said. When I implemented it (I did an internship in Bologna this winter), I basically copied how IBEX was handling it, but made a mistake. 
I don't know to which "hardware errors" conditions the specs refer to. PMP exceptions for example are already handled synchronously. Also in PULPissimo RI5CY (nor IBEX) use the NMI line as of now.

Silabs-ArjanB, 2020-05-18:
Hi @abettati , no problem at all, would you be okay if we remove NMI all together and add it in a future core once this functionality has been standardized by the REISC-V organization. I put the proposal to remove the pin in https://github.com/openhwgroup/core-v-docs/issues/50 (which is part of a bigger effort to get the documentation updated, see https://github.com/openhwgroup/core-v-docs/issues/45)

abettati, 2020-05-18:
Hi @Silabs-ArjanB looks good to me! It surely makes sens to postpone this. Think we can close this:)

abettati, 2020-05-18:
@Silabs-ArjanB I read your proposal for the update to CLINTXpulp and I wish we had discussed this while I was working on it, It could have been great for me.
By the way, I'm writing my master thesis on the current CLINTXpulp implementation, I could provide it in a few days as I finish it. Maybe you could find there some useful material for the new docs.

Silabs-ArjanB, 2020-05-27:
@abettati I would love to see your master thesis indeed. I am sure we can benefit from it with respect to our documentation

MikeOpenHWGroup, 2020-05-29:
@abettati, I would also be interested in your thesis (and I suspect others would as well).  Will it be publicly available?

abettati, 2020-05-29:
Hi @MikeOpenHWGroup, sure it will be in a few days and I already sent a complete draft to @davideschiavone last week.

abettati, 2020-05-31:
@MikeOpenHWGroup @Silabs-ArjanB  I have a quick question in the meantime: are you also planning a compliance test for the interrupts? I see that RI5CY ran tests from [here](https://github.com/riscv/riscv-tests), but I don't think they check IRQ behavior. 
In my thesis i only implemented a simple c firmware ([interrupt_test](https://github.com/openhwgroup/cv32e40p/tree/master/tb/core/interrupt_test)) to check that interrupts are correctly executed, but I didn't use anything official from RISC-V.

MikeOpenHWGroup, 2020-05-31:
Hi @abettati.  The short answer to your question is , "yes, we are planning a complete verification of the interrupts for CV32E40(P)."   This would include all interrupt features specified by the ISA plus any CV32E40(P) specific behaviors.

This effort has not started, and I am actively recruiting for a Contributor for this feature.  If you, or any of your colleagues are interested, please reach out to me (mike@openhwgroup.org) and I would be happy to discuss our plans.

By the way, verification of the CV32E40(P) is executed using the [core-v-verif](https://github.com/openhwgroup/core-v-verif) project.  If you are interested in knowing more, the [Verification Strategy](https://core-v-docs-verif-strat.readthedocs.io/en/latest/) is a good place to start.

Silabs-ArjanB, 2020-06-09:
Hi @abettati The NMI related functionality has been removed all together. Could you please close this issue?

closed 2020-06-09 by abettati