openhwfoundation/cv32e40p #338: clock gating information required

alexnetterville, 2020-05-14:
Hi,
We have been running our RISC-V formal proofkit on the cv32e40p. 
Depending on what we do with clock_en_i at the top level, we are prone to seeing failures around load instructions.
We are seeking some clarity around the expected behaviours of the system that surrounds the cv32e40p so as to eradicate incorrect environmental setup.
When we run all proofs with clock_en_i tied off to 1'b1, no proofs fail.
When we run with no constraint on clock_en_i, we see a case where load instructions can fail.
The following description is from a counter example we have had out of one of the formal tools. We intend to file a bug later if one is needed on this, but this description is given just to start the discussion we require here. 
We have seen that the LSU pipeline has its clock gated when the RVALID is returning load data. When you look at both NS and CS of the main state machine in the LSU, CS never updates to take in the read data via the WAIT_RVALID state. As a result, the returned data is missed. 
We wanted to know whether or not the core can indicate that it is active or not, and if so what are the necessary implications for system level behaviour, such as the RVALID response in the above example. 
Ideally, the behaviour of an upstream response shouldn't be an issue as the core will know to wait for it, but since there is an issue that has exposed itself here it either indicates there is something more our testbench needs to know, or there is a bug in the core that means the LSU is able to prematurely go into a dormant state and miss returned data, rather than postpone a shut down of the clock knowing that the memory will return its data.
Can you clarify any system level requirements we may require, or let us know if you think there may be an issue requiring further investigation? If the latter is the case, we can provide a VCD.

alexnetterville, 2020-05-14:
[axiomise_cv32e40p_misaligned_load_word_clock_gating_anomaly.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/4630302/axiomise_cv32e40p_misaligned_load_word_clock_gating_anomaly.vcd.gz)

Silabs-ArjanB, 2020-05-15:
Hi @alexnetterville , thank you for the detailed report. I do not know the full answer to your question, but I hope I can make some useful remarks. First of all, the fact that clock_en_i is used at all must mean that the top level parameter PULP_CLUSTER must be set to its (non-default) value 1 (otherwise clock_en_i should have been ignored completely by the CV32E40P). Of course it is great that you also check this configuration. I fully agree with your observation: By specification the rvalid signals (data_rvalid_i and instr_rvalid_i) are 1 cycle long pulses on clk_i, so the core will miss these pulses if we allow arbitrary behavior on clock_en_i when PULP_CLUSTER = 1. Maybe there is a undocumented use restriction on clock_en_i or some overlying protocol potentially using the core_busy_o signal. There is (currently) no way for the environment to really know if the internal clock in the load/store unit is running or not. @davideschiavone could you please describe if there are any restrictions on when clock_en_i is allowed to change?

davideschiavone, 2020-05-15:
Hello Alex, thanks for you report.

As Arjan said, consider the core to be used always with PULP_CLUSTER equal to 0 as that parameter is only used with special conditions. It has been implemented to use the core in the PULP Cluster (as the name said) where there exists multiple instances of the core (4 or 8 usually) running digital signal processing applications efficiently.
As they need to synchronize, custom instructions (p.elw) have been implemented to make LOAD operations on the EVENT UNIT , which implements barriers, semaphores, etc in an efficient way (the cost of a load instruction only).

This special load to the Even Unit mmap registers together with the core_busy signal, tell the even unit that the core is not busy and ready to go to sleep. Only in that case the event unit drives the clock enable low, saving the cluster dynamic power. Once the core are ready to start again (let's say, the last core met the barrier), the clock enable is driven high.

So the clock enables does care about the valid in a indirect way, i.e., by driving the core_busy signal

I hope I helped you understanding better

Silabs-ArjanB, 2020-05-25:
Hi @alexnetterville , I understood you would rerun the proofs with PULP_CLUSTER==0. Do you have a trace for that?

alexnetterville, 2020-05-26:
Hi Arjan,

Yes, I have rerun the proofs using PULP_CLUSTER == 0, as you guys have suggested, and we now aren’t seeing the failure around the internal clock being gated. We will be presenting our results soon of proofs we have been doing, and it would be good to share those with you once our work is complete. We are almost done with RV32I!

Thank you guys so much for your assistance and your suggestions around the core parameters. I won’t hesitate to get back to you guys if I have anything further.

Alex

Silabs-ArjanB, 2020-05-26:
Hi @alexnetterville , good to hear that the issue is not present for PULP_CLUSTER == 0 and looking forward to your presentation. Could you close this specific issue as we are now tracking the documentation task related to the PULP Cluster interface (that will describe the rules for clock_en_i) via https://github.com/openhwgroup/core-v-docs/issues/49 ?

alexnetterville, 2020-05-26:
Hi Arjan,

I will do this right away. We are looking forward to any documentation updates that may come our way regarding this, so that we can maybe explore a bit more the PULP_CLUSTER == 1 configuration once the rules are available.Will this documentation include an explanation of what each of the parameters does?

Alex

alexnetterville, 2020-05-26:
I am going to close this issue since - as above - the issue is not presenting itself with pulp_cluster == 0.

closed 2020-05-26 by alexnetterville

Silabs-ArjanB, 2020-05-26:
Yes, we have a task/issue to properly document all parameters and pins: https://github.com/openhwgroup/core-v-docs/issues/45