openhwfoundation/cv32e40p #355: Correct license headers (if needed)

Silabs-ArjanB, 2020-05-27:
@MikeOpenHWGroup suggested that we should pre-pend the OpenHW "solder-pad" license header (leaving the existing header in place).

As part of this task the following questions should be answered/discussed here first:

- Should license headers be pre-pended or should existing headers be updated/replaced?
- Should all files have headers (e.g. many files do not have any license header (e.g. certain testbench files or tcl scripts)

MikeOpenHWGroup, 2020-05-27:
I've assigned this issue to myself and will check with the Eclipse people to get their opinion and report back.

MikeOpenHWGroup, 2020-05-27:
I got the following from Wayne Beaton, Director of Open Source Projects at the Eclipse Foundation:

1. We should not change the headers of content unless we either are or have permission from the copyright holders to do so.
2. All files should have license headers unless they cannot (e.g. JSON files do not support comments).

An additional point is that the Eclipse Foundation will be conducting an Intellectual Property Due Diligence review of all of the code/files in our GitHub repos.  I have no timeline on that, but it seems we will get specific feedback on topics such as licensing and comment headers as part of that process.

Given point 1 above and the pending review, I recommend we do nothing at this time.   Marking the issue as "Resolved" and assigning it back to @Silabs-ArjanB  for closure (or re-opening, as he sees fit).

Silabs-ArjanB, 2020-05-28:
We do not comply to item 2) above. @MikeOpenHWGroup can you please remove the label resolved (I don't have permission to do so)

MikeOpenHWGroup, 2020-05-28:
That is correct, item 2 says we do not have comment headers on everything.  How we are supposed to handle this (and other related items) will come out of Eclipse's Intellectual Property Due Diligence review, and its possible said review will require us to un-do any changes we may make as a result of this issue.

I've removed the label, but still recommend we close this issue.

Silabs-ArjanB, 2020-05-28:
OKay; closing it now and then we open a new ticket if so required by the outcome of Eclipse's Intellectual Property Due Diligence review.

closed 2020-05-28 by Silabs-ArjanB