openhwfoundation/cv32e40p #357: Beginner Question: CSR access generates Illegal instruction

saumeister, 2020-06-01:
Hi everyone,

I'm currently trying to port some bench-marking programs to run on the core. I have encountered some problems with the timer section in CoreMark where I try to read  CSR mcycle during start and stop measurements.

The timer code used:
```
void start_time(void) { 
	uint32_t mcyclel;
        asm volatile ("csrr %0,mcycle"  : "=r" (mcyclel));
        stop_time_val = mcyclel;
}
```
```
void stop_time(void) {
	uint32_t mcyclel;
        asm volatile ("csrr %0,mcycle"  : "=r" (mcyclel));
        stop_time_val = mcyclel;
}
```

This result in the following output in the Questa Sim Transcript:
```
[CORE] Core settings: PULP_SECURE =  0, N_PMP_ENTRIES = 16, N_PMP_CFG  4
BENCHMARK TIMER STARTED
573865ns: Illegal instruction (core 0) at PC 0x00002b46:
illegal instruction exception handler entered
 
594665ns: Illegal instruction (core 0) at PC 0x00002b64:
illegal instruction exception handler entered
 
out of bounds read from 06153f2f
** Fatal: Assertion error.
    Time: 615960 ns  Scope: tb_top.riscv_wrapper_i.ram_i.read_mux File: mm_ram.sv Line: 373
** Note: Data structure takes 90129808 bytes of memory
          Process time 10.82 seconds
          $finish    : mm_ram.sv(373)
    Time: 615960 ns  Iteration: 2  Instance: /tb_top/riscv_wrapper_i/ram_i
```

Do you have to enable the CSR registers by yourself? I did read in the ibex manual that all CSR registers are enabled by default after reset but could not find the same statement regarding this core.

I would appreciate if somebody have any pointers if I'm on the right path or if I should do a different approach to make program execution time measurements for bench-marking purposes.

Silabs-ArjanB, 2020-06-02:
Hi @saumeister , the performance counters have very recently been changed from the 'PULP specific counters' to RISC-V compliant counters from the Counters section in https://riscv.org/specifications/isa-spec-pdf/. The original PCMR and PCERR CSRs no longer exist and the RISC-V compliant counters will default to be inhibited (via mcountinhibit).

Unfortunately the related documentation change has not been merged in yet, but you can see the updated performance counter docs at https://github.com/openhwgroup/core-v-docs/pull/59

saumeister, 2020-06-02:
Hi @Silabs-ArjanB, 

Thank you for taking you time to answer. I will make sure to use the RTL from the master branch and look into the new RISC-V compliant counters and give it another try now.

saumeister, 2020-06-04:
Hi,

Thanks for the help. It worked out fine after updating the RTL and some small code modifications.

closed 2020-06-04 by saumeister