openhwfoundation/cv32e40p #364: Failing compliance tests for cv32e40p core

PauliusMorku, 2020-06-12:
Hello,
it might be a known and irrelevant problem but I still decided to open a new issue. There are three riscv compliance tests that are failing:
I-NOP-01.S - # Test part A2 - general test, only 5 cycles are executed
I-SB-01.S - # Test part E2 - test war hazard (address register)
I-SH-01.S - # Test part E2 - test war hazard (address register)

Assertions for these tests are triggered on core (testbench based) and uvmt_cv32 (UVM based) verification environments. I used QuestaSim 10.7b, and master branch HEAD commits of cv32e40p and core-v-verif (same behaviour was noticed with older commits).
I would like to know if these tests are deprecated or there are some problems with cv32e40p core itself. 
Thank you

MikeOpenHWGroup, 2020-06-12:
Hi @PauliusMorku, thanks for raising the issue.  This is a known problem and will be addressed shortly.  Just FYI, the updates will not be added to this repository, rather they will be added to the [core-v-verif](https://github.com/openhwgroup/core-v-verif) project.

Here is the current status of this issue:
1. As you report, it is a known issue with the test(s).
2. Updated compliance tests from the foundation are available and ready to be installed at core-v-verif.
3. Installation of the updated tests has been gated by toolchain and "board support package" issues.
4. The BSP issues are resolved and so getting the toolchain and testcases straightened out should be completed next week.

I've assigned this issue to myself and will report back.   Thanks again!

Silabs-ArjanB, 2020-09-24:
Hi @PauliusMorku The RISC-V compliance tests are now part of the 'ci check' on core-v-verif. You can follow the discussion on the tests here https://github.com/openhwgroup/core-v-verif/issues/253 instead. Could you please close this (more or less) duplicate issue and address any Compliance test questions to core-v-verif?

davideschiavone, 2020-10-21:
Can we close this issue?

MikeOpenHWGroup, 2020-10-21:
From my perspective, yes we can close this issue.

Silabs-ArjanB, 2020-10-28:
Closing due to inactivity and under the assumption that the RISC-V Compliance Tests are made to work in the core-v-verif repos (where it is being tracked by e.g. https://github.com/openhwgroup/core-v-verif/issues/302, https://github.com/openhwgroup/core-v-verif/issues/308)

closed 2020-10-28 by Silabs-ArjanB