openhwfoundation/cv32e40p #365: bit manipulation: RTL does not match spec for p.extract permutations and p.clb

GTumbush, 2020-06-12:
We are writing directed self-checking tests for the pulp extensions and started with the bit manipulation instructions.  We are finding numerous cases in which the RTL does not match the documentation.

The first suspected bug is with: 
p.extract rD, rs1, Is3, Is2
The definition is:
Sext((rs1 & ((1 << Is3) – 1) << Is2) >> Is2)* Note: Is3 + Is2 must be <= 32

For instruction p.extract x18, x20, 8, 0  where x20=0x69879573 
rD=x18, rs1=x20=0x69879573, ls3=8, and ls2=0

The operation is:  Sext((0x69879573 & ((1 << 8) – 1) << 0) >> 0) so Sext(0x73)=0x73

The RTL is calculating 0xFFFFFF73

The next suspected bug is for instruction: p.extract x18, x20, 16, 2  where x20=0x78729209
We expect the x18=0xFFFFA482 but the RTL is calculating 0xA482

It seems the RTL is sign extending using the wrong sign bit.

documentation: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html#bit-manipulation-operations
RTL: cv32e40p repo, master branch, hash 101f455

To run this testcase with Xcelium:
git clone -b pulp_extension_bugs https://github.com/GTumbush/core-v-verif.git 
export CV32E40P_PKG=<path to your RI5CY core>
export SIMULATOR=xrun
export USE_ISS=NO
cd cv32/sim/uvmt_cv32/
make pulp_bit_manipulation

Note, you must have RISCV and RISCV_EXE_PREFIX set to a compiler which supports pulp extensions.

A screen shot of the first bug is:
[image]

Anytime X15 increments in a suspected bug.  See attached spreadsheet for a description of each bug.

[Bug listing for pulp bit manip.xlsx](https://github.com/openhwgroup/cv32e40p/files/4773181/Bug.listing.for.pulp.bit.manip.xlsx)

GTumbush, 2020-06-12:
@MikeOpenHWGroup 
Hi Mike, can you please assign this to @davideschiavone  I do not have the ability. @jm4rtin if you are interested in working on these bugs please state so.

GTumbush, 2020-06-12:
Note that the test has no $finish criteria so will run forever.  The relevant instructions are done executing in 3us.

GTumbush, 2020-06-23:
Updated test with exit code from riscv_ebreak_test_0 so test completes.

jm4rtin, 2020-06-26:
@davideschiavone Do you have a human readable description of the bit manipulation instructions? Perhaps when they were originally defined? It would be good to add one into the documentation so someone doesn't have to decipher the bitwise operation to figure out how it works.

**p.extract:** rD = Sext((rs1 & ((1 << Is3) – 1) << Is2) >> Is2)
Extract **ls3 bits** starting at bit position ls2 from rs1 and store the sign extended result into rD.

**p.extractu:** rD = Zext((rs1 & ((1 << Is3) – 1) << Is2) >> Is2)
Extract **ls3 bits** starting at bit position ls2 from rs1 and store the zero extended result into rD.

**p.insert:** rD = (rD & ~(rs1[Is3:0] <<Is2)) | (rs1[Is3:0] << Is2)
Insert **ls3+1 least significant bits** (staring at bit 0) into rs1 starting at bit position ls2 and store the result into rD.

These operands are not actually symmetric (operand for how many bits to extract or insert) in their definition but the hardware implementation is. It's odd for the documentation to state that they are not symmetric.

davideschiavone, 2020-06-30:
Hi @jm4rtin , I just pushed on the core-v-docs the new description of the bit manipulation. It's hopefully more readable, actually, it would be good if you give feedbacks :)

jm4rtin, 2020-07-01:
@davideschiavone When looking at **p.clb** it appears to produce the leading bits -1 and not the leading bits. This also returns 31 for the case of all 1's. I would expect it to return the leading bits like the documentation says but I would like to check with you before I make a change to this instruction. Was this done because clb_result-1 is going into both the result register and the div/rem unit?

Also, we would like some clarification on **p.bitrev**. It notes that ls3 can be either 1, 2 or 3 but the case in the **ALU is looking for 0, 1, or 2**. Also, framing the description as part of an FFT doesn't describe the instruction very well. It looks like it is shifting the input value and then reversing in groups of 1, 2 or 3 bits (and in the "radix=8" case it ignores the upper 2 most significant bits)?

Digging deeper I have some examples:
```
p.bitrev x18, x20, 0, 4 (radix-2)

in:  0xC64A5933 11000110010010100101100100110011
out: 0x0CC9A526 00001100110010011010010100100110

Swap pattern:
A B C D E F G H . . . . . . . . . . . . . . . . . . . . . . . .
0 1 1 0 0 1 0 0 1 0 1 0 0 1 0 1 1 0 0 1 0 0 1 1 0 0 1 1 0 0 0 0
. . . . . . . . . . . . . . . . . . . . . . . . H G F E D C B A
0 0 0 0 1 1 0 0 1 1 0 0 1 0 0 1 1 0 1 0 0 1 0 1 0 0 1 0 0 1 1 0
```

```
p.bitrev x18, x20, 1, 4 (radix-4)

in:  0xC64A5933 11000110010010100101100100110011
out: 0x0CC65A19 00001100110001100101101000011001

Swap pattern:
A  B  C  D  E  F  G  H  I  J  K  L  M  N  O  P
01 10 01 00 10 10 01 01 10 01 00 11 00 11 00 00
P  O  N  M  L  K  J  I  H  G  F  E  D  C  B  A
00 00 11 00 11 00 01 10 01 01 10 10 00 01 10 01
```

```
p.bitrev x18, x20, 2, 4 (radix-8)

in:  0xC64A5933 11000110010010100101100100110011
out: 0x216B244B 00100001011010110010010001001011

Swap pattern:
A   B   C   D   E   F   G   H   I   J
011 001 001 010 010 110 010 011 001 100 00
   J   I   H   G   F   E   D   C   B   A
00 100 001 011 010 110 010 010 001 001 011
```

Should I just add this to the documentation?

davideschiavone, 2020-07-13:
Hi @jm4rtin , 

you are right. The documentation is wrong. I checked the test we wrote for it and indeed it's Leading bit - 1

https://github.com/davideschiavone/riscv_pulpissimo_test/blob/master/testCnt/testCnt.c

For the brev you are also right. The documentation should be:

so **radix2 = 0, radix4 = 1, radix8 = 2**

This instruction works with the lenght of a signal as well, that should be a multiple of 2, 4 or 8.

so if your signal has 2^16 samples, its indexes are represented in 16 bits:

Example, radix2:

0000 0000 0000 0000 ABCD FGHJ KLMN PQRS

0000 0000 0000 0000 SRQP NMLK JHGF DCBA

Example, radix4: (4^8)

0000 0000 0000 0000 AB CD FG HJ KL MN PQ RS

0000 0000 0000 0000 RS PQ MN KL HJ FG CD AB

In radix8 you cannot write 2^16 as 8^something

so let's say you have a signal 512 (8^3) , so it representable in 9bits

0000 0000 0000 0000 0000 000 ABC DFG HJK

0000 0000 0000 0000 0000 000 HJK DFG ABC

so to group data by 3bits you can have at most 30bits (2^30 samples), that's why the last 2 are necessary tight to 0.

Of course you can submit a PR to the core-v-docs to specify these cases.

sorry again for the delay

best
Davide

jm4rtin, 2020-07-13:
@davideschiavone Just to be clear, on all cases the documentation should be updated. No RTL change needed, correct?

davideschiavone, 2020-07-13:
yes correct. The tests assembly/C tests we wrote and the RTL are coherent. We are just not good in documenting it :)

jm4rtin, 2020-07-13:
@davideschiavone Okay I'll submit a PR to fix the documentation (unless you beat me to it). Are the tests you looked at in the repository or something internal at ETH?

davideschiavone, 2020-07-13:
Most of the tests are here:
https://github.com/davideschiavone/riscv_pulpissimo_test

I will update the repo with the Complex one but I don't want to support such repository. The verification will take care of better one. We can use this one to clarify such cases thoug :)

closed 2020-08-10 by davideschiavone