openhwfoundation/cv32e40p #371: Path from *_rvalid_i to performance counter

Silabs-ArjanB, 2020-06-17:
Combinatorial paths exist from the *_rvalid_i inputs to mhpmcounter_q[3]. The rvalid signals are typically late arriving inputs the the stated performance counter uses a 64-bit incrementer.

Would it be okay to count all events 1 cycle later (thereby preventing the performance counters from contributing to critical paths)?

MikeOpenHWGroup, 2020-06-17:
Its "probably" OK, as long as its documented in the user manual.  Verificiation does not yet have a Vplan captured for performance counters, so I do not know if we will need to check these on-the-fly or at the end-of-sim.  If its end-of-sim, then this change will not impact verification.  If we need on-the-fly, then the TB will need to "know" this.

davideschiavone, 2020-07-07:
I think the error is here:
```

        if( cnt_idx == 0)
          // mcycle = mhpmcounter[0] : count every cycle (if not inhibited)
          mhpmcounter_n[cnt_idx] = **mhpmcounter_n**[cnt_idx] + 1;

        else if(cnt_idx == 2)
          // minstret = mhpmcounter[2]  : count every retired instruction (if not inhibited)
          mhpmcounter_n[cnt_idx] = **mhpmcounter_n**[cnt_idx] + hpm_events[1];

        else if( (cnt_idx>2) && (cnt_idx<(NUM_MHPMCOUNTERS+3)))
          // add +1 if any event is enabled and active
          mhpmcounter_n[cnt_idx] = mhpmcounter_q[cnt_idx] +
                                   |(hpm_events & mhpmevent_q[cnt_idx][NUM_HPM_EVENTS-1:0]) ;
```

it should be  

mhpmcounter_n[cnt_idx] = **mhpmcounter_q**[cnt_idx] + 1;

mhpmcounter_n[cnt_idx] = **mhpmcounter_q**[cnt_idx] + hpm_events[1];

Can you please fix it ? @silabs-PaulZ

davideschiavone, 2020-07-07:
I reopened the issue as this should be first checked with synthesis, for the moment I fixed that typo

Silabs-ArjanB, 2020-07-07:
I will run a synthesis to check; thanks!

Silabs-ArjanB, 2020-07-08:
The combinatorial paths still exist, but the unwanted long paths are gone, so the fix looks good. (Fixing the combinatorial path all together would most likely involve changing the cv32e40p_controller, which is not considered worth the effort)

PR #397 (2020-07-07): :bug: fix #371
Fix #371
commit 4dfa9995 (2020-07-07): :bug: fix #371
commit 8a3345cd (2020-07-07): Merge pull request #397 from davideschiavone/fix371

:bug: fix #371
closed 2020-07-08 by Silabs-ArjanB