openhwfoundation/cv32e40p #378: Vectored interrupt changes ?

Imperas, 2020-06-19:
Hi All,

Previously we had a testcase which would setup vectored interrupts by writing 0x8101 to the MCAUSE register. This should setup the BASE address as 0x8100 in vectored mode.
If we then generated an external interrupt (code = 11) setting the MCAUSE register to 0x8000000b indicating (Interrupt / Machine External Interrupt) this would then vector to address 0x812c 

I have taken code at hash 748abbf, and now when the Interrupt occurs the MCAUSE is set to 0x00000001 indicating (Exception / Instruction access fault) 

any pointer to what has happened here ?
Thx
Lee

Silabs-ArjanB, 2020-06-19:
I am a bit confused as to why you think you can set the vector base address and mode via MCAUSE. Do you mean to say MTVEC?

Imperas, 2020-06-19:
> I am a bit confused as to why you think you can set the vector base address and mode via MCAUSE. Do you mean to say MTVEC?

Yes, apologies, I will edit the original post
Thx
Lee

Silabs-ArjanB, 2020-06-19:
Maybe the memory model of the test becnh can not yet deal with the OBI interface and returns the wrong instructions. Is core-v-verif already pointing to 748abbf or later? I can have a look at this and check the memory model.

Silabs-ArjanB, 2020-06-19:
Can you point me to your test case and how to run it?

MikeOpenHWGroup, 2020-06-19:
Have a look at https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/Common.mk to see where core-v-verif is pointing (its currently 9135b2f, which is well back).  The next pr will get us to 101f455 which is still being the OBI changes.

Imperas, 2020-06-19:
@GTumbush - Greg can you suggest a way that we can easily provide an example of the interrupt issue you discovered ?

Silabs-ArjanB, 2020-06-19:
Maybe easiest to get https://github.com/openhwgroup/core-v-verif/pull/118 merged, then we are all in sync again.

GTumbush, 2020-06-22:
I'm sorry, I don't have any way to trigger an interrupt in the OpenHW testbench.

Silabs-ArjanB, 2020-06-23:
Hi @GTumbush I recreated (and fixed) the issue you encountered  with tb/core/interrupt_test. It should be fixed once https://github.com/openhwgroup/cv32e40p/pull/383 has been merged.

Silabs-ArjanB, 2020-07-14:
Hi @GTumbush , #383 has been merged. Could you please close this issue if you agree it has been fixed?

GTumbush, 2020-07-14:
@Silabs-ArjanB   Imperas filed this issue but I can check the fix. What command line did you use to run interrupt_test? I tried:
bash-4.2$ make interrupt_test
make: *** No rule to make target `interrupt_test'.  Stop.

Silabs-ArjanB, 2020-07-14:
Hi @GTumbush The 'test' was renamed. You can run 'make interrupt-vsim-run'

GTumbush, 2020-07-14:
@Silabs-ArjanB 
Sorry, still not compiling.
bash-4.2$ make interrupt-vsim-run
make: *** No rule to make target `interrupt-vsim-run'.  Stop.

Silabs-ArjanB, 2020-07-16:
Hi @GTumbush No problem. I have no idea why that doesn't work for you. Are you on the most recent repos? The makefile target is defined here and works on a clean repos for me (using Questa): https://github.com/openhwgroup/cv32e40p/blob/3fcdeef90561d8df2f89a92d9566121fd1b2eb3f/example_tb/core/Makefile#L243

eroom1966, 2020-07-16:
Hi @Silabs-ArjanB @GTumbush 
I am just getting back up to speed here

I see that the target names includes vsim as part of the name, does this imply that these targets are specific to this particular RTL simulator?
Previously we had this working on all major RTL simulators
Thx
Lee

Silabs-ArjanB, 2020-07-16:
Hi @eroom1966 I greatly simplified the Makefile by removing many targets (e.g. compliance test suite) and simulator options. We were having way too much duplication of code between core-v-verif and this repos. This repos will have very minimal test case and tool support. Also we will be moving to use FuseSoc; I do not really know how that will work, but maybe that is the time to enable other tools in an easy manner.

Silabs-ArjanB, 2020-08-19:
Hi @Imperas Please let us know if there is still an issue here (we think it has been fixed). If we do not hear back then this issue will be closed automatically a week from now.

eroom1966, 2020-08-19:
@Silabs-ArjanB 
@strichmo (FYI)

Hi, I took a look at this again, not sure what target to run, but I think it is interrupt ?

so I run
make SIMULATOR=xrun interrupt
this gets stuck in a tight loop, I presume waiting for an interrupt which never occurs

I should add I am working on the rm_update branch, maybe this is on a specific branch ?

Thx
Lee

eroom1966, 2020-08-19:
Further findings
I am running the interrupt test under a software debugger and trying to understand the test here
```
    // Enable all mie (need to store)
    ie_mask32_std = 0xFFFFFFFF;
    ie_mask32_1   = 0xFFFFFFFF;

    asm volatile("csrw 0x304, %[ie_mask32_std]"
                  : : [ie_mask32_std] "r" (ie_mask32_std));
    asm volatile("csrw 0x7D0, %[ie_mask32_1]"
                  : : [ie_mask32_1] "r" (ie_mask32_1));
```

The CSR @0x7D0 as far as I can tell does not exist, and this traps to an illegal instruction - is that intentional ?
Is this historical ?

Silabs-ArjanB, 2020-08-19:
Hi @eroom1966 I think you are looking at old code; the 0x7d0 CSR should no longer be mentioned in the software nor the RTL (if you use today's master branch).

Silabs-ArjanB, 2020-09-14:
Hi @Imperas We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question or comment remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-09-14 by Silabs-ArjanB