openhwfoundation/cv32e40p #379: Cannot gate clock for WFI when PULP_CLUSTER = 1

Silabs-ArjanB, 2020-06-22:
In the configuration with PULP_CLUSTER = 1 the internal architectural clock gate (i.e. core_clock_gate_i) cannot be gated.

Once the WFI has been hit, the core_busy_o signal will go low as intended. In order to gate the internal clock the environment will have to lower clock_en_i as the final gating condition is determined as follows:

  assign clock_en      = PULP_CLUSTER ? clock_en_i | core_busy_o : irq_pending | debug_req_i | core_busy_o;

Once both clock_en_i and core_busy_o are low a pending and locally enabled interrupt will not be signaled via core_busy_o as clock_en will remain low and the core will not become 'busy'. The environment could have kept clock_en_i high to prevent this situation, but that would leave the internal clocks running during a WFI, which is not desired. 

Likely irq_pending (and debug_req_i ?) need to factor into clock_en for PULP_CLUSTER = 1 as well.

davideschiavone, 2020-06-23:
The WFI should not be used when the PULP_CLUSTER parameter is set to 1. 
I suggest sto implement the WFI as a NOP in that case and to create a new documentation issue to do so

davideschiavone, 2020-06-23:
Separatly in another response here, the even unit handles the interrupts and debug requests, i.e., it receives them and forwards them to the core after waking it up. That's why the are not part of the formula of clock_en. When the core uses the PULP_CLUSTER=1 it relies complitly on the event unit for power managment.

Silabs-ArjanB, 2020-06-23:
Okay, that makes sense. So the CV32E40P with PULP_CLUSTER = 1 will never receive an interrupt nor a debug request while pulp_clock_en_i = 0, right?

Silabs-ArjanB, 2020-06-23:
Can we define the requirements for PULP_CLUSTER = 1 as follows?

- WFI shall be implemented as NOP (such that it does not impact core_sleep_o)
- While pulp_clock_en_i = 0, the environment must guarantee that instr_gnt_i = 0, instr_rvalid_i = 0, data_gnt_i = 0, data_rvalid_i = 0, debug_req_i = 0, irq_i[] = 0 (not sure if we should demand this independent of local/global interrupt enable bits or not)
- pulp_clock_en_i = 0 is only allowed when core_sleep_o = 1

closed 2020-07-07 by Silabs-ArjanB