openhwfoundation/cv32e40p #381: Non-standard CSRs

jm4rtin, 2020-06-22:
There is several non-standard CSRs in the core:
* lpstart0
* lpend0
* lpcount0
* lpstart1
* lpend1
* lpcount1
* privlv
* uhartid
* fprec

Hardware loop CSRs are only present if this feature is enabled (disabled in the current configuration). The ```uhartid``` CSR should only be present if user mode is enabled (disabled in the current configuration). It is actually ***still present*** even without user mode. The ```fprec``` CSR is only present if the FPU is enabled (disabled in the current configuration).

That leaves the ```privlv``` CSR still in the current core configuration which needs to be addressed.

If these CSRs stay they should be added to the ISS. Another option is to just remove them if they serve no useful purpose.

Silabs-ArjanB, 2020-06-22:
Hi @jm4rtin Thank you for this ticket. @davideschiavone Could you please comment on uhartid; we left it available even in a M-mode only machine as this is how it was on RI5CY and we expected that whoever is using this wanted it to be available from any mode (so having it available in User mode to make it future proof, for the same reasoning that the FPU CSRs are User Mode CSRs even though we don't support User mode).

On behalf of my company I can state that we do not need any of the unconditional non-standard CSRs. They are all there because they existed in RI5CY. @davideschiavone should we maybe make them optional on XPULP?

Silabs-ArjanB, 2020-06-22:
By the way, also MIP1 and MIE1 are non-standard. All the non-standard CSRs are clearly documented in https://github.com/openhwgroup/core-v-docs/pull/106

jm4rtin, 2020-06-22:
Ah yes sorry, I forgot those. We removed them from our version 😄 

As a short summary of my discussion with Mike on mattermost:
My preference would be to remove ```privlv``` as it should not be needed. The privilege level can be read by the debugger from the ```dsrc``` CSR. Any other usage would be for software running on the core; however, that creates a virtualization hole. The intention in the ISA was to not provide this information to running code. See some discussions on this topic:
* https://groups.google.com/a/groups.riscv.org/forum/#!topic/isa-dev/JeqieDbIpO4
* https://forums.sifive.com/t/how-to-determine-the-current-execution-privilege-mode/2823

It could stick around but in my opinion it should have a good reason to stay.

Why is ```uhartid``` needed? Regardless it is redundant to have it in machine mode where you can read ```mhardid``` instead and causes a CSR to be added to the ISS.

Silabs-ArjanB, 2020-06-22:
Hi @jm4rtin Fully agreed with your remark about the virtualization hole; that is also why we do not like to have this. @davideschiavone needs to weigh in on this. Maybe we can find a compromise where we use a parameter to guard uhartid and privlv; we do not need either of them either. I guess/assume the argument for the uhartid is that in that case your code would remain unchanged if you would later add User mode to a core.

davideschiavone, 2020-07-01:
Hi @jm4rtin , thanks for addressing this.

The `uhartid `should be readable only if USER_MODE is implemented in my opinion. 
So it can stay there but probably should be parametrizable and not exposed to systems where only M-Mode is implemented (as by default)

For what concern the `privlv` CSR, I let Germain @haugoug replies on that as this was a SW requirement for PULP systems.

Thanks
Davide

haugoug, 2020-07-01:
I agree, these 2 CSRs, prilv and uhartid, are specific to pulp, they should not be visible. We will still need them, just having them enabled with a parameter would be perfect.

Silabs-ArjanB, 2020-07-01:
I have a proposal to remove some custom CSRs and to move other behind a (default true) parameter XPULP.

Proposal is to have three PULP related parameters:

- PULP_CLUSTER: Remains as is
- PULP_ZFINX: Remains as is
- XPULP: 
  - All PULP instructions excluding p.elw
  - All hardware loop related functionality (PULP_HWLP parameter will be removed)
  - All custom CSRs

    lpstart0
    lpend0
    lpcount0
    lpstart1
    lpend1
    lpcount1
    fprec

The following CSRs will either be only included if XPULP=1 or be removed all together (my preference):

    privlv
    uhartid

The following CSRs will be removed all together:

    mie1 (I discussed an alternative mechanism together with Davide to provide 32 interrupts without vector table holes)
    mip1 (I discussed an alternative mechanism together with Davide to provide 32 interrupts without vector table holes)

jm4rtin, 2020-07-01:
That would make the core either 100% standard or include everything custom. It would be handy to have such a powerful parameter to turn it all off but it doesn't give you much control over individual features like the hardware loop or subsets of custom instructions. Is this a proposal to turn it all off for CV32E40 (the one after CV32E40P)?

Silabs-ArjanB, 2020-07-02:
Hi @jm4rtin If you have a proposal for finer-grained parameter control then that is welcome of course. We just do not want to go too fine-grained as then the number of combinations will get large causing difficulties/overhead for verification. The current proposal for CV32E40 can be found here https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/CV32E40P_and%20CV32E40_Features_Parameters.pdf; the idea is to only  support the PULP extensions for which there is no official RISC-V alternative, not support hardware loop, not support PULP cluster. For the PULP extensions that would remain the idea is to pull them out of the core and simply make a generic interface available such that people can add their own instructions externally (and we could then give an example for some of the PULP instructions).

Silabs-ArjanB, 2020-07-02:
Hi @haugoug Could you please explain the scenario in which you need privlvl and uhartid in a machine-mode only core as CV32E40P? Also, are they only needed in case PULP_CLUSTER = 1 (in which case we could maybe make them conditional on that parameter)?

Silabs-ArjanB, 2020-07-06:
Hi @haugoug @jm4rtin Could you please comment on the above. I am now ready to issue a pull request for the following:

- PULP_CLUSTER: Remains as is
- PULP_ZFINX: Remains as is
- PULP_XPULP: All PULP ISA extensions including hardware loop, but excluding p.elw. This parameter also guards all custom CSRs (including prvlvl and uhartid, but excluding mip1/mie which were agreed to be removed all together)

jm4rtin, 2020-07-06:
I am fine with adding the XPULP parameter. I still prefer `privlv` and `uhartid` be removed all together but it's not critical. Adding some finer grained local parameters might be useful to block out specific extensions:
* **PULP_LDST** - Post-increment load & store, register-register load & store
* **PULP_HWLP** - This one already exists
* **PULP_BITMANIP** - Bit manipulation ALU extensions
* **PULP_ALUEXT** - All other ALU extensions not already listed (or this and bitmanip could be combined into one parameter)
* **PULP_MAC** - Multiply-accumulate
* **PULP_SMID** - SIMD instructions

Another good argument for blocking out some of the PULP instructions is that leaving all of them in the core consumes all the custom instruction encoding regions and some reserved regions defined within the RISC-V specification. For example, perhaps some users may want the load/store and ALU extensions but not the MAC & SIMD extensions leaving space for their own custom instructions.

But this may deserve an issue of it's own.

haugoug, 2020-07-06:
Hi @Silabs-ArjanB, for some chips with small internal memory, we don't need virtualization, just a bit of isolation without code duplication, where such instructions are quite useful.
I am also fine with your proposal about the parameters, and also the one with finer control.

Silabs-ArjanB, 2020-07-07:
Hi @haugoug , @jm4rtin Thank you for your feedback. I decided to go for the more coarse gain control (i.e. PULP_XPULP) although I appreciate the fine grained proposal above. Reason to stick to the coarse grain control (for now) is mainly to prevent the verification burden implied by introduction of separate parameters (implying that all combinations need to be verified); also there would be an implied tool support, which is not the case (as in the PULP tool chain the custom ISA extensions cannot be enabled/disabled individually). Now basically instead of having PULP_HWLP we have PULP_XPULP which is just wider in scope than the original parameter, but which does not add more combinations of parameter options to be verified. The corresponding pull request can be found at https://github.com/openhwgroup/cv32e40p/pull/398. I will make a pull request for the associated documentation update next.

Silabs-ArjanB, 2020-07-16:
Hi @jm4rtin Can this issue be closed now? If not, please let us know what still needs to be done to achieve closure.

closed 2020-07-16 by jm4rtin