openhwfoundation/cv32e40p #382: tracking load instructions help request

alexnetterville, 2020-06-22:
We have a checker that tracks load instructions entering the core decode and emitting on the dreq interface. 
Since we have taken the latest core over what we had previously been using (which was a few months old), this mechanism is now firing.  when the tracked instruction is emerging onto the interface, the data values disagree.
I have enclosed a VCD of a filing trace for you to examine. 
The real question is how to track loads interspersed with many other instructions (including stores) from issue to dreq transaction completion, and what should the increment and decrement clauses be from a high level architectural perspective (given this core's microarchitectural behaviour and choices about transaction break up).

We are aware of the statement in the manual that says writes require an rvalid to complete them.

[tracking_error_on_aligned_load_word.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/4814516/tracking_error_on_aligned_load_word.vcd.gz)

Silabs-ArjanB, 2020-06-22:
Hi @alexnetterville Thank you for filing this report and for the waves file. Can you please point me to a failing/firing assertion or a value anywhere in the many signals in the waveform that you suspect is wrong? (I don't see assertions in the vcd)

With respect to transaction break up the following will happen:

A load/store transaction is broken into 2 load/store transaction for the following scenarios:

- load/store of a word to a non-word-aligned address
- load/store of a halfword crossing a word address boundary

In both cases the transfer corresponding to the lowest address is done first. I see this is not sufficiently spelled out in the User Manual and I will add this info there right away.

I hope this answers your question.

alexnetterville, 2020-06-22:
Hi Arjan

You’re welcome.

I think the main signals of interest are as follows:

u_isa.load_word_smart_tracker.tracking_counter

u_isa.load_word_smart_tracker.tracking_counter_enable

And compare it to the instruction stream and the data interface activity. Maybe something has gone out of alignment somewhere. This might provide a start.

Thank you for the information regarding the transaction break up. This is what we have surmised. We are curious though - why is an rvalid needed to complete a write? How is the system expected to produce this rvalid? And how many rvalid are expected to confirm a misaligned write that has two accesses?

Cheers, Alex

Silabs-ArjanB, 2020-06-22:
Hi @alexnetterville 

>> why is an rvalid needed to complete a write? 

This is a way in which the environment can hold off completion of a store; in the future a bus error signal can also become associated with rvalid (as it was on earlier versions of RI5CY).

>> How is the system expected to produce this rvalid? 

One or more cycles after every granted request (i.e. rising clk edge with data_req_o and data_gnt_i both 1) a 1 cycle rvalid pulse needs to be generated.

>> And how many rvalid are expected to confirm a misaligned write that has two accesses?

2 (the bus protocol applies for each bus access in the same way).

I have no way of debugging the internal signals that you added. The bus u_isa.load_word_smart_tracker.tracking_counter and u_isa.load_word_smart_tracker.tracking_counter_enable look fine to me in the sense that the count goes up to 2 as there are two loads and then it decrements by one for each rvalid received. You have to be more explicit in pointing to something that is not as you expect otherwise we cannot help. To me it looks like we just received a database of signals without any indication which signal behaves different than you expect at some specific point in time.

alexnetterville, 2020-06-22:
Hi Arjan.

Thanks for your reply. The intention of the VCD and this ticket is to get to the bottom of what the requirements are for tracking the load transactions through the core, with some waves provided for you to look at if necessary. I agree that if I had a bug, I would give you exact signals and times etc. When I believe I have a bug to file, that is what I will provide to you, absolutely.

On to your answers:

1) RVALID needed for Writes: Why isn’t GNT just used for delaying the write?

2) producing rvalid: The thrust of my question is more around whether it is appropriate to place a requirement on the SoC to acknowledge a write as if it were a read. Is it the case that a SoC is expected to implement logic local to the processor and away from the interconnect fabric to create the response, or is the interconnect expected to produce this? If so, how does an interconnect like AXI integrate with this? AXI has a response channel for writes (B channel) that will acknowledge writes separately to the AXI R channel for reads, so is extra muxing logic required outside of the interconnect? How do you ensure that rvalids for reads aren’t accidentally acknowledging writes, containing valid data that the write instruction shouldn’t just be dropping on the ground?

3) number of RVALIDs for misaligned writes: Thanks for acknowledging this.

Alex

Silabs-ArjanB, 2020-06-23:
With respect to 1)

Grant delays the address phase; rvalid delays the response phase. Both are useful, in particular if we would associate e.g. an error signal with the rvalid. Having rvalid also for writes makes the protocol orthogonal as well. 

With respect to 2)

>>appropriate to place a requirement on the SoC to acknowledge a write as if it were a read. 

The protocol is indeed identical for reads and writes.

>>Is it the case that a SoC is expected to implement logic local to the processor and away from the interconnect fabric to create the response, or is the interconnect expected to produce this? 

The end-point slave produces grant and rvalid, but of course they need to pass through interconnect fabric as well (which might also introduce buffering, pipelining, further waits, default grants, etc.). OBI is a point-to-point protocol only concerned about one master connected  directly to one slave.

>> If so, how does an interconnect like AXI integrate with this? AXI has a response channel for writes (B channel) that will acknowledge writes separately to the AXI R channel for reads, so is extra muxing logic required outside of the interconnect? 

You would need AXI-OBI adapters. The AXI response channels would need to be merged on the single OBI response channel. The OBI address channel would need to be demuxed to the appropriate AXI address channel.

>>How do you ensure that rvalids for reads aren’t accidentally acknowledging writes, containing valid data that the write instruction shouldn’t just be dropping on the ground?

This is because OBI is an in-order protocol; having rvalid for writes actually makes this easy to implement for a bus interconnect. See R-6 of https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/OBI-v1.0.pdf

alexnetterville, 2020-06-23:
Hi Arjan,

Thank you for these answers. That clarifies for us what your requirements are on the environment, and is very helpful. You could perhaps consider adding some of this information to the core integration chapter:
https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html  .

I would also like to ask if the manual could be updated here https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/load_store_unit.html , since according to the specification:
data_rvalid_i   input   data_rdata_i holds valid data when data_rvalid_i is high. This signal will be high for exactly one cycle per request.
This does not mention that it is also high for writes. I would also change the wording to reflect that it isn’t always high for valid rdata, because the core will ignore rdata if the response was for a write. Could the section below that starts “Note that data_rvalid_i must also be set when a write was performed” be referenced from the above line in the table (or maybe better to add a summary line to the table in line with what that paragraph says so that write responses are mentioned in the table with some detail)? Otherwise, it looks like a contradiction by omission in the specification between the table and the paragraph below.

Many thanks,

Alex

https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html

Silabs-ArjanB, 2020-06-23:
Hi @alexnetterville I updated the documentation according to the above (see https://github.com/openhwgroup/core-v-docs/pull/113). Could you please close this issue?

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it has been answered at least 7 days ago. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB