openhwfoundation/cv32e40p #384: Do not make glob imports of packages in the compilation unit scope unless necessary

bluewww, 2020-06-23:
There are imports like this 
https://github.com/openhwgroup/cv32e40p/blob/8e22c994198ad08f4cacf239d56d98e1cc25627b/rtl/cv32e40p_core.sv#L30-L38

which makes all the variables visible in the whole compilation unit. This can cause issues when this codebase is compiled in the same compilation unit as another module that has a variable with the same name as in one of those packages. We saw this issue before e.g. when using vcs versus vsim. In vsim we had one compilation unit per file, but vcs had one compilation unit for the whole design.

The better approach is doing the import inside the module or just after the module name (same effect)

i.e. 
```verilog
module cv32e40p_core import cv32e40p_defines::*;
#(
  parameter PULP_HWLP           =  0,                   // Hardware Loop (not supported yet; will be supported)
  parameter PULP_CLUSTER        =  0,

[...]
```

Silabs-ArjanB, 2020-06-24:
Hi @bluewww Fully agreed; could you actually make a pull request for these changes yourself?

bluewww, 2020-06-29:
Sorry I didn't get around to make those changes, but I see you already worked on it. Thanks.

Silabs-ArjanB, 2020-07-14:
Hi @bluewww The corresponding pull request has been merged. Could you please close this issue if you agree it has been fixed or let us know what issue still exist?

bluewww, 2020-07-14:
@Silabs-ArjanB thanks looks correct

closed 2020-07-14 by bluewww