openhwfoundation/cv32e40p #388: Debug spec - Trigger registers access

mp-17, 2020-06-29:
Hello everyone,

I saw that the decoder of cv32e40p allows access to the **trigger registers** only when the core is in _debug mode_.
**RISC-V Debug Specification** states that they can be accessed in machine mode as well:

> The  trigger  registers  are  only  accessible  in  machine  and  Debug  Mode  to  prevent  untrusted  user code from causing entry into Debug Mode without the OS's permission.

(RISC-V External Debug SupportVersion 0.13.2d5029366d59e8563c08b6b9435f82573b603e48e, page. 48, par. 5.2)

I've not checked yet if there are also other registers with access restriction problems.

Thank you and best regards,
Matteo

jm4rtin, 2020-06-29:
This feature is not implemented in the current design. The debug trigger CSRs can be accessed in machine mode if dmode=0; however, dmode is tied to 1 in the design. The description of dmode:

> 0: Both Debug and M-mode can write the tdata registers at the selected tselect.
> 1: Only Debug Mode can write the tdata registers at the selected tselect. Writes from other modes are ignored.

It should probably not throw an illegal instruction exception on access to these CSRs and instead ignore the write (as stated).

silabs-PaulZ, 2020-07-07:
I agree with the change request to adhere to the spec.
To summarize:
- Keep the Mode tied  1 (Only allow Debug Mode to write)
- In non-debug Machine mode
  - Reads of trigger CSRs will return actual register values
  - Writes will be ignored

Access will no longer cause an illegal instruction. Reads will always reflect CSR actual values (as opposed to always returning zero value).
 
Note that CV32E40P only supports Machine mode. When we add User mode (in CV32E40), I would assume that access to the trigger CSRs will result in an illegal instruction.

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-14 by Silabs-ArjanB