openhwfoundation/cv32e40p #401: MTVEC interrupt vector default value

silabs-PaulZ, 2020-07-08:
The default value of the MTVEC, interrupt and exception vector address, is set to 0. If an exception occurs before SW has reconfigured MTVEC to the proper location then the processor will be in a deadlock situation. 

We should add the mtvec_addr_i port, similar to boot_addr_i, that defines the default value of the MTVEC CSR value. This allows the user to customize the default boot and MTVEC locations together. MTVEC will still adhere to the specification where it can be changed via a CSR write.

MikeOpenHWGroup, 2020-07-08:
Good catch @silabs-PaulZ.  Just to play the devil's advocate here... it is common practice for s/w to initializes the GPRs as part of its init code.  If interrupts are all masked by default and the init code updates MTVEC, will this not prevent the deadlock?

Full disclosure - as a hardware guy, I am loath to push an initialization requirement onto software.  Nevertheless its a common practice.

silabs-PaulZ, 2020-07-08:
Indeed the interrupts will be masked until software enables them (mstatus.mie), after mtvec update. This change is to address a non interrupt exception. Granted, this would be caused by poor programming.

bluewww, 2020-07-08:
Normally the first thing you do in a `crt0.S` is to set `gp`, `sp` and `mtvec`. The spec doesn't say anything about the reset value of mtvec so I think this proposed change is fine

davideschiavone, 2020-07-08:
This is very dangerous. It works only if the input is tighy with a CONSTANT value.
Otherwise the synthesizer will try to put the Q of some flip-flop value in the RESET state of the MTVEC flip flops.
I propose:

1. Either we use a PARAMETER (safe and nice but yet another parameter)

2. Or we build an FSM that writes the mtvec_addr_i after reset into mtvec (ugly)

silabs-PaulZ, 2020-07-08:
Hi @davideschiavone ,

I want to have the MTVEC CSR updated with mtvec_addr_i after the first cycle out of reset (Option 2). This will be similar to the IBEX solution, except that IBEX resets the MTVEC to the same value as the boot address.
The IBEX solution can be found here:
https://github.com/lowRISC/ibex/blob/3d29e5174c2c1826bfaf8ed1a0800e23858475af/rtl/ibex_cs_registers.sv#L451
https://github.com/lowRISC/ibex/blob/3d29e5174c2c1826bfaf8ed1a0800e23858475af/rtl/ibex_if_stage.sv#L155

jm4rtin, 2020-07-08:
I think changing the reset value is unnecessary and complicates the design and verification. Just set it as the first instruction of your crt0. If you *have* to make it configurable I wouldn't recommend a parameter as this precludes changing it via metal options. The IBEX approach is okay.

Silabs-ArjanB, 2020-07-09:
I am also in favor of this fix if it is done in a similar manner as on Ibex.

silabs-PaulZ, 2020-07-09:
Hi @davideschiavone , @jm4rtin , @Silabs-ArjanB ,

The IBEX solution also defaults the interrupts to vectored mode. CV32E40P is currently defaulted to direct mode. I plan to make the change match the IBEX solution (default vectored mode).

Silabs-ArjanB, 2020-07-10:
Hi @silabs-PaulZ Why not simply make a **mtvec_i** port instead so that all implemented bits of mtvec can be initialized? That would be even more general and easier to explain.

silabs-PaulZ, 2020-07-10:
Hi @Silabs-ArjanB ,

I was thinking that we only wanted to change the address. The mode bit does not affect the main purpose of this fix which is to provide the default address for non-interrupt synchronous exceptions say from the first instruction being corrupt. Interrupts are default disabled, so SW would need to configure this mode bit appropriately before enabling them. Moreover, the mode is dependent on the SW implementation. My general thinking is that default addresses should be configurable (boot,dm_halt,dm_exception_halt,mtvec) as these have a dependency on the HW SOC address mapping, but modes should always come up the same (to ease verification and SW expectation).

I’m OK with it either way but slightly prefer to just have the address be configurable.

Silabs-ArjanB, 2020-07-10:
Hi @silabs-PaulZ Fully agreed; forgot that the mode only affects interrupts and not exceptions

closed 2020-07-23 by Silabs-ArjanB