openhwfoundation/cv32e40p #404: Missed debug_req_i pulse

Silabs-ArjanB, 2020-07-10:
A debug_req_i pulse can be missed if it comes in during the part of the enable process of CV32E40P in which the interal clock is gated off.

MikeOpenHWGroup, 2020-07-10:
Hi @Silabs-ArjanB - good catch.  I assume you caught that with inspection, rather than simulation.  What would a testbench need to do to ensure it could reproduce this issue.  Are there similar issues that verification should be aware of?

Silabs-ArjanB, 2020-07-10:
This was the result of a directed formal test I did aimed at finding issues with the architected clock gate. The statement that a 1 cycle pulse on debug_req_i is enough is part of the User Manual.

MikeOpenHWGroup, 2020-07-10:
Nice.   I've created [a Vplan issue](https://github.com/openhwgroup/core-v-docs/issues/128) for @silabs-oysteink to ensure we capture that in the Vplan.

I suspect there is more to this issue: your formal test looks for one cycles pulses on debug_req_i, but the issue seems to be related to "part of the enable process".  What is that?

Silabs-ArjanB, 2020-07-10:
With the enable process I mean the assertion of fetch_enable_i as described in https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html. Certain other signals depend on that signal (as described in the same section) and clock gating depends on it as well (as also described but awaiting a pull request)

closed 2020-09-14 by Silabs-ArjanB