openhwfoundation/cv32e40p #407: pv.cplxmul.(r/i)( /.div2/div4/div8) incorrect functionality

DavidM-EMUS, 2020-07-13:
The compressed folder below has a test in it for all complex number SIMD instructions (pv.subrotmj( /.div2/.div4/.div8), pv.cplxconj, pv.cplxmul.(r/i)( /.div2/,div4/.div8)). The test is rather simple in nature, wherein rD is always x19, rs1 is always x17, and rs2 is always x18. rD, rs1, and rs2 are loaded with randomized values, and then the pulp instruction being tested is run. Additionally, x20 is always loaded with the expected output value from the tested instruction, and x15 is used as an error counter to keep track of the number of incorrect comparisons (x20!=x18). The final value of x15 at the end of the test is the number of errors.

[pulp_vectorial_complex.zip](https://github.com/openhwgroup/cv32e40p/files/4913914/pulp_vectorial_complex.zip)

None of the instructions are recognised by the toolchain. so .word (hex value) is used, but more importantly, all 8 pv.cplxmul._ instructions do not perform as expected. According to the documentation at
https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html#simd-complex-number-operations
complex numbers are set up such that rD[15:0] is the real component, and rD[31:16] is the imaginary component.
Thus, pv.cplxmul.r( /.div2/.div4/.div8) rD, rs1, rs2 should have results in the following format:
rD[15:0] = (((rs1[15:0]*rs2[15:0]) - (rs1[31:16]*rs2[31:16]))>>(15/16/17/18), rD[31:16] = rD[31:16],
and pv.cplxmul.i should have results in the following format: rD[15:0] = rD[15:0], rD[31:16] = ((rs1[15:0]*rs2[31:16]) + (rs1[31:16]*rs2[15:0]))>>(15/16/17/18). Unfortunately, as far as I can tell, the instructions have been flipped. This can be seen in the test as the output from the cplxmul instructions consistently has the wrong half-word being modified, and the half-word that is modified has the result of the opposite instruction's math exporession. In other words, cplxmul.r performs the way cplxmul.i( /.div2/.div4/.div8) should according to the documentation, and cplxmul.i performs the way cplxmul.r( /.div2/.div4/.div8) should.

davideschiavone, 2020-07-14:
Thanks a lot @DavidM-EMUS , I will try to give it a look asap

davideschiavone, 2020-08-10:
Hello @DavidM-EMUS , sorry for the late reply.
Actually you found two bugs. One in the tracer, and one in the documentation.

I checked the original specs, and I mistankely inverted the bit 25, both in the documentation and in the tracer.
But the RTL is doing the right thing, i.e. if bit 25 is 1, then it is an imaginary instruction.

So I will fix the tracer as soon as PR #436 is merged (to avoid a lot of mergin operations). Whereas I fix now the documentation.

Thanks
Davide

davideschiavone, 2020-08-10:
This PR fixes the documentation https://github.com/openhwgroup/core-v-docs/pull/160

davideschiavone, 2020-09-03:
@DavidM-EMUS, this issue has been fixed and merged to master. Can you confirm that it works please?

closed 2020-09-03 by davideschiavone

DavidM-EMUS, 2020-09-03:
Just ran the test that exercises those instructions, and everything looks great. Thanks for taking care of this!

commit 429f485b (2020-09-03): Merge pull request #479 from davideschiavone/fix407

:bug: fix #407
commit 7437d1c7 (2020-09-03): :bug: fix #407