openhwfoundation/cv32e40p #411: Reinstate the TRACER

MikeOpenHWGroup, 2020-07-14:
The tracer was removed in commit 8a626b9.   The tracer supports step-and-compare with the verification environment's reference model which makes it a **must have** for verification.   This will gate all current and future pull requests against core-v-verif to sync the with RTL including https://github.com/openhwgroup/core-v-verif/pull/113 and https://github.com/openhwgroup/core-v-verif/pull/148.

Silabs-ArjanB, 2020-07-14:
Hi @MikeOpenHWGroup We are working on this; the tracer will get bound to the core as opposed to being instantiated inside the core.

MikeOpenHWGroup, 2020-07-14:
OK.  Do you have a time estimate on this?   The updates for mtvec and debug and the updated parameters are all gated by this.

Silabs-ArjanB, 2020-07-15:
The tracer has been integrated already in https://github.com/openhwgroup/core-v-verif/pull/113, so there is no need to revert anything.

If we do the following in core-v-verif we are all in sync again:

Step 1) Updates the CV32E40P_HASH as requested in https://github.com/openhwgroup/core-v-verif/pull/148 and merge this pull request
Step 2) Make the requested updates in https://github.com/openhwgroup/core-v-verif/pull/113 and merge this pull request

davideschiavone, 2020-07-16:
Let's keep this issue open until we have a wrapper as suggested in https://github.com/openhwgroup/core-v-verif/pull/148
I assign this task to myself

davideschiavone, 2020-07-16:
@Silabs-ArjanB , @MikeOpenHWGroup , shall we modify the core-v-verif to use such wrapper?

MikeOpenHWGroup, 2020-07-16:
Thanks for the offer @davideschiavone.   I am taking care of that now.

closed 2020-07-23 by Silabs-ArjanB