openhwfoundation/cv32e40p #415: Missing "fpnew_pkg" parameter

YuliaR-2006, 2020-07-16:
Hi, 
I'm trying to synthesize the RTL code and I see that the  "fpnew_pkg" parameter which is used in 'rtl/include/cv32e40p_pkg.sv' , 'rtl/cv32e40p_ex_stage.sv', 'rtl/cv32e40p_decoder.sv' etc...  is not defined anywhere.

jm4rtin, 2020-07-16:
See: https://github.com/pulp-platform/fpnew

The FPU is normally parameterized out so if you leave it out of the core you really only need the package.

Silabs-ArjanB, 2020-07-17:
Hi @YuliaR-2006 You can type 'make deps' in cv32e40p to clone that repos. If this solves your issue can you please close this ticket?

Silabs-ArjanB, 2020-08-19:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-08-19 by Silabs-ArjanB