openhwfoundation/cv32e40p #418: Simultaneous request of debug, trigger, and interrupt

silabs-PaulZ, 2020-07-17:
The controller state machine has a bug where it cannot handle an interrupt and either debug or trigger being requested at the same time.  The FSM is encoded in a one-hot manner, expecting only one of these conditions to be active. Moreover, an assertion was previously written to catch this overlap but has been inadvertently disabled.

https://github.com/openhwgroup/cv32e40p/blob/1607d8b675864db1aa013364fd4444a665331830/rtl/cv32e40p_controller.sv#L427-L451

The proposal to fix this is to replace the unique case with if then else; higher prioirity on debug,trigger, interrupt, then other. 

Also, we should remove the invalid assertion.
  [assert property (
    @(posedge clk) (~('0 & irq_req_ctrl_i)) ) else $warning("Both dbg_req_i and irq_req_ctrl_i are active");](https://github.com/openhwgroup/cv32e40p/blob/1607d8b675864db1aa013364fd4444a665331830/rtl/cv32e40p_controller.sv#L1152)

git hash: 916d92afc6bbc27b7ab65c503043982e1b6e3ab0

[image]

davideschiavone, 2020-07-17:
Hi @silabs-PaulZ , thanks for brining this up.
The proposal is realistic and I agree with it. I would probably give priority to interrupt first as usually more timing sensitive for the application than the debug. But those are corner cases anyway. So I don't have strong opinions on it.

silabs-PaulZ, 2020-07-17:
Hi @davideschiavone , the irq and debug will result in a halt in the decode. If the irq is first, then after the irq flush and irq fetch, the (irq) decode will be halted due to debug req. So, it probably would be better to simply take debug_req first since it will essence block the irq irregardless. But alas, it is minor, we are entering debug which will significantly impair the irq response (if not block it all together during debug_mode).

Silabs-ArjanB, 2020-07-18:
@silabs-PaulZ Good catch! Shouldn't formal have caught the fact that the above 'unique case' can be violated? If not, then we should probably add equivalent assertions for all 'unique case' statements written in above manner.

closed 2020-09-04 by silabs-PaulZ