openhwfoundation/cv32e40p #426: how to build cv32e40p encoding.h , encoding.h as shown follow

ncf125, 2020-07-21:
#define clear_csr(reg, bit) ({ unsigned long __tmp; \
  if (__builtin_constant_p(bit) && (bit) < 32) \
    asm volatile ("csrrc %0, " #reg ", %1" : "=r"(__tmp) : "i"(bit)); \
  else \
    asm volatile ("csrrc %0, " #reg ", %1" : "=r"(__tmp) : "r"(bit)); \
  __tmp; })

#define rdtime() read_csr(time)
#define rdcycle() read_csr(cycle)
#define rdinstret() read_csr(instret)

#endif

#endif

#endif

#endif
/* Automatically generated by parse-opcodes */
#ifndef RISCV_ENCODING_H
#define RISCV_ENCODING_H
#define MATCH_BEQ 0x63
#define MASK_BEQ  0x707f
#define MATCH_BNE 0x1063
#define MASK_BNE  0x707f
#define MATCH_BLT 0x4063
#define MASK_BLT  0x707f
#define MATCH_BGE 0x5063
#define MASK_BGE  0x707f
#define MATCH_BLTU 0x6063
#define CSR_MCYCLE 0xb00
#define CSR_MTIME 0xf01
#define CSR_MINSTRET 0xf02

Silabs-ArjanB, 2020-07-21:
Hi @ncf125 , I do not understand the actual issue/question you have. Can you please clarify?

ncf125, 2020-07-21:
The CSR register address is defined when c is compiled.   We need a  encoding.h file that will define these. Previously RISCV was generated automatically.I don't know how to generate such a file. Thanks

Silabs-ArjanB, 2020-07-21:
Hi @ncf125 None of the example programs in the cv32e40p use/require encoding.h as far as I can see. Where did you find that file being used and where was it generated originally?

ncf125, 2020-07-21:
https://github.com/pulp-platform/riscv-opcodes?files=1   I just find it .  Let me study it

closed 2020-07-22 by ncf125