openhwfoundation/cv32e40p #427: Security hole in hardware loop, sketch of a possible mitigation

sorear, 2020-07-21:
## The problem

There appears to be no way to stop a U-mode program from setting `lpstart0` to the address of an instruction in M-mode and redirecting M-mode execution.  This can also be done with the dedicated instructions if M-mode and U-mode code are close enough in memory.  Clearing hardware loops in trap entry is not sufficient as the U-mode program can set `lpstart0` to an early instruction in the trap handler.

The security issue only exists for cores that claim to provide any security, however some possible solutions have software compatibility implications.

## Disfavored alternatives

The simplest fix for the protection hole and the most consistent with the handling of other mode bits in the architecture would be to either replicate the hardware loop state for every privilege level, or add save registers for the hardware loop state at every non-U privilege level and atomically save and clear, or restore, the state on all traps to or from non-U privilege levels. However, this would not be compatible with existing assembly code that uses the same lpXXX CSRs to configure and save/restore the loop state and regardless of privilege level.  Additionally, doubling the size of the hardware loop state (or even just replicating the end registers) is likely unattractive for the smallest M+U configurations.

The obvious alternative is to tag hardware loops with the privilege level they were configured at and only use loops configured at the current level.  Most ways to do this either overtly break existing code, introduce subtle vulnerabilities in existing code, require involved modifications to the trap process, are not compatible with more than two privilege levels, and/or introduce virtualization holes.  I believe the approach below could work.

Implementing just the active bit part of this proposal and not the CSR would have the security and software compatibility advantages of the full proposal but would hinder future architectural evolution (it would not be compatible with S-mode or any form of virtualization).

## Simplified description of the subset of the proposal for M+U hardware:

Allocate two new CSRs, "lppriv" in the HRW space and "slppriv" in the SRW space.  Both CSRs behave identically in M and U modes so they can be decoded together.

Add a two bit protection mode register for each implemented hardware loop. This records the privilege level at which the loop was last configured. A loop is active if it was configured at the current privilege level.

The loop privilege CSRs are divided into eight 4-bit sections, supporting up to eight hardware loops.  (More than this is straightforward but probably not useful.)  The description below assumes one hardware loop is implemented; it is to be intepreted as describing hardware replicated for each loop.

The functionality of reading and writing the lpend0, lppriv, and slppriv CSRs is as described in pseudo-C below.  Implicit writes by the lp.setup family of instructions and implicit reads in the fetch unit are affected in the same way. Since valid instruction addresses never have the low bit set, lpend implict reads which set the low bit can never match an instruction.  If M-mode is the only implemented protection level, this reduces to lppriv and slppriv reading as hardcoded zero registers.

Note that lppriv and slppriv have WARL behavior and map all unsupported values to INACT\_FORCE.  Note that there is no change to trap returns and a loop can have a protection level _higher_ than the current privilege mode; this represents erroneous software behavior but is architecturally deterministic.

If there is absolutely no desire to support Xpulp on MMU cores ever slppriv can be dropped, saving about two gates on M-only and M+U cores.

    typedef enum { M = 0, U, INACT_FORCE, INACT_OTHER } prot_state;

    /* state for each hw loop */
    prot_state loop_priv;
    long end_addr;

    /* execution privilege level mapped into the same namespace */
    extern prot_state current_priv;
    extern _Bool mode_implemented(prot_state mode);

    long rd_lpend() {
        /* Loops are active for the current mode only. */
        if (loop_priv == current_priv) {
            return end_addr;
        } else {
            return end_addr | 1;
        }
    }

    void wr_lpend(long end) {
        /* Setting an active loop sets it for the current mode.
           Setting an inactive loop in M-mode is considered restoring a U-mode
           loop, and is implemented as such to support naive M and U modes.
           Otherwise, behavior is priv-independent for virtualization. */
        if ((end & 1) && mode_implemented(U)) {
            loop_priv = (current_priv == M) ? U : INACT_FORCE;
        } else {
            loop_priv = current_priv;
        }
        end_addr = end & ~1;
    }

    prot_state rd_lppriv(_Bool is_slppriv) {
        if (current_priv == U) {
            trap();
        }
        return loop_priv;
    }

    void wr_lppriv(prot_state priv, _Bool is_slppriv) {
        if (current_priv == U) {
            trap();
[... 96 lines cut ...]

    abbreviations: IO (OTHER), IF (FORCE), IOV (OTHER_VS)

    GUEST                 (classical)       (H-extension)
    LPPRIV   GUEST MODE   M   S   VS  U     S/U  VS/VU

    M        HOST LPPRIV  U   IO  IO  IO    IO   IO
    S                     IO  U   IO  IO    VS   IO
    VS                    IO  IO  U   IO    IO   VS
    U                     IO  IO  IO  U     U    U
    IO                    IO  IO  IO  IO    IOV  IO
    IF                    IF  IF  IF  IF    IF   IF
    IOV                   IO  IO  IO  IO    IO   IOV

## Properties of the proposal

* All code which runs at a single privilege level and only writes valid addresses to lpendN is unaffected by the proposal.
* Trap handlers which do not use hardware loops cannot affect or be affected by loops configured at a lower privilege level, as they will be inactive during execution at the higher privilege level.
* If a M-mode trap handler saves and restores lpendN but fails to save and restore lppriv, all U-mode and inactive loops will become active U-mode loops when restored.  This is a change of U-mode state and thus to be avoided, but does NOT allow control flow hijacking and does NOT break currently valid U-mode code (which does not inactivate loops).  Thus, M+U mode programs where all code is currently valid (only writes aligned addresses to lpendN, ignores
  values read from lpendN except to save and restore) will be protected against control flow hijacking but not otherwise affected.
* S-mode trap handlers which save and restore lpendN but fail to save and restore (s)lppriv will leave all loops in the INACT\_FORCE state; this will NOT allow control flow hijacking but WILL cause otherwise valid U-mode code to malfunction.
* The only overhead of implementing the proposal for M-mode-only cores is two new hardwired zero CSRs and two instructions in trap handlers that save loop state.
* Overhead of nontrivial implementations is two or three flops per implemented hardware loop and several dozen gates (TBD) on the existing lpend and new lppriv read and write paths.  No change is required to trap handling and no new trap conditions (other than regular privilege checks for a new CSR) are introduced.
* Only one bit in lpendN is used, so it does not preclude support for 2-byte aligned loops.
* Virtualization is possible merely by emulation of CSRs and adjustment of the state across emulated traps and trap returns.

Silabs-ArjanB, 2020-07-22:
Hi @sorear, thank you very much for reporting this issue and obviously spending a lot of time to think it through and to propose a solution. Even though CV32E40P only supports user mode, I indeed agree that we need to look ahead and make fixes also in a machine mode only implementation as well to be able to write forward compatible software. I am not really in the position to comment on the desired fix (I will leave that to @davideschiavone in this case).

pascalgouedo, 2023-08-23:
Hi @Silabs-ArjanB & @davideschiavone 

As CV32E40P is not going to support anything else than Machine mode and not used as a Secure either, can we close this issue?

Silabs-ArjanB, 2023-08-23:
Okay with me

closed 2023-08-28 by davideschiavone