openhwfoundation/cv32e40p #429: HPM counter logic 'possibly' not implemented correctly

silabs-PaulZ, 2020-07-22:
@davideschiavone ,

This issue was found by inspecting the RTL. It also serves as a reminder to write a test for HPM. The RTL looks to increment the jump, branch, and branch_taken based on a signal asserted during the decode phase (e.g. jump_i) along with a pipelined control signal (id_valid_q). 

https://github.com/openhwgroup/cv32e40p/blob/1607d8b675864db1aa013364fd4444a665331830/rtl/cv32e40p_cs_registers.sv#L1352-L1363

Perhaps this should be replaced with the non-pipelined instruction retired condition (e.g. jump_i & id_valid_i & is_decoding_i).
hash: 8a626b912ba5491c9fb384a4e498554ca579cae3

Silabs-ArjanB, 2020-09-11:
I can confirm (at least) two bugs in above code:

1) The performance counter registers software writes do not take effect on the correct instruction boundaries (the fix for this is the the insertion of flush states as is done for many other CSRs)
2) The id_valid_q signal is not necessarily aligned with jump_i as is shown in below waveform (this can be fixed with qualifying jump_i differently). (Maybe similar bugs exist for the other id_valid_q uses)

  45026ns            4498 00000788 00000013 nop                                 
  45036ns            4499 0000078c 00000013 nop                                 
  45046ns            4500 00000790 00012203 lw               x4, 0(x2)            x4=xxxxxxxx  x2:0000f9f0  PA:0000f9f0
  45066ns            4502 00000794 2cc00267 jalr             x4, x0, 716          x4=00000798
  45076ns            4503 000002cc 03010113 addi             x2, x2, 48           x2=0000fa20  x2:0000f9f0

[image]

Silabs-ArjanB, 2020-09-14:
Formal shows that id_valid_q is implied by branch_i, so following will be simplified by removing id_valid_q:

assign hpm_events[8]  = branch_i                   & id_valid_q;       // nr of branches (conditional) 
assign hpm_events[9]  = branch_i & branch_taken_i  & id_valid_q;       // nr of taken branches (conditional) 

Following will be written more intuitively:

 assign hpm_events[3]  = jr_stall_i & id_valid_q;                       // nr of jump register hazards 

as 'jump_i && inst_ret && jr_stall_q' // stall cycle precedes jump

Silabs-ArjanB, 2020-09-15:
Fix done in https://github.com/openhwgroup/cv32e40p/pull/504

Silabs-ArjanB, 2020-09-23:
Hi @silabs-PaulZ Can this be closed now (the related merge has been done)?

closed 2020-09-23 by silabs-PaulZ