openhwfoundation/cv32e40p #431: Update marchid

Silabs-ArjanB, 2020-07-23:
marchid is currently implemented as a 'read 0' CSR, which is legal. 

It would be better to:

- Use the value as defined for RI5CY at https://github.com/riscv/riscv-isa-manual/blob/master/marchid.md
- Make a pull request at https://github.com/riscv/riscv-isa-manual/blob/master/marchid.md to change RI5CY into CV32E40P

MikeOpenHWGroup, 2020-07-23:
This is being a little nit-picky, but I do not think the read value of zero is strictly illegal.  section 3.1.13 of the priviledged spec reads:
```
This register must be readable in any implementation, but a value of 0 can be returned to indicate the field is not implemented.
```
In the same sub-section it goes on to say,
```
Open-source project architecture IDs are allocated globally by the RISC-V Foundation
```
There does not seem to be a requirement that a given core should actually use this as its marchid.

Having sad that, I agree with @Silabs-ArjanB's suggestions and think we should do both.

Silabs-ArjanB, 2020-07-23:
Agreed, as said above, the current implementation is legal as well.

Silabs-ArjanB, 2020-08-19:
Sent email to the maintainer, kgf@iis.ee.ethz.ch, to make the following pull request:

-	Project name: CV32E40P
-	Maintainers: OpenHW
-	Point of contact: Davide Schiavone <davide@openhwgroup.org>
-	Architecture ID: 4
-	Project URL: https://github.com/openhwgroup/cv32e40p

Silabs-ArjanB, 2020-08-31:
Changed into Doc issue as the RTL update has been performed already; now just awaiting the update of the documentation at https://github.com/riscv/riscv-isa-manual/blob/master/marchid.md

davideschiavone, 2020-09-10:
A PR is pending here:

https://github.com/riscv/riscv-isa-manual/pull/578

davideschiavone, 2020-09-12:
> 
> 
> A PR is pending here:
> 
> [riscv/riscv-isa-manual#578](https://github.com/riscv/riscv-isa-manual/pull/578)

The PR has been merged. We can close this issue

closed 2020-09-12 by Silabs-ArjanB