openhwfoundation/cv32e40p #434: taken bge followed by div takes 36 cycles

GTumbush, 2020-07-29:
**Component:RTL**: 

For specific combinations of bge followed by div the flush of the div takes 36 clock cycles. The branch is taken.
Other combinations of branch/div/mul/rem and operands may reproduce this issue.  I will enumerate a few below. 

### Steps to Reproduce
1. git hash: 2fe68a93c27db0bb514d567f79d3ba54045b51f6
2. git clone -b add_riscv_rand_instr_test_0 https://github.com/GTumbush/core-v-verif
3. cd cv32/sim/uvmt_cv32
4. make custom CUSTOM_PROG=riscv_branch_div

For:   
bge   a2, zero, fast_exit
div        a3, a6, a6
[image]

For:   
bge   a2, zero, fast_exit
div        a3, a6, **s1**
[image]

For:   
bge   a2, zero, fast_exit
**nop**
div        a3, a6, a6
[image]

GTumbush, 2020-08-05:
@MikeOpenHWGroup can you please assign this bug? I cannot.

MikeOpenHWGroup, 2020-08-06:
Hi @davideschiavone.  Do you have time to look at this one?  It is urgent for em-micro.

GTumbush, 2020-09-01:
Cloned Silabs-ArjanB:ArjanB_bge_div and verified as fixed.  
 
For:
bge a2, zero, fast_exit
div a3, a6, a6
[image]

Other scenarios are similar.

GTumbush, 2020-09-01:
Should not close until merged.

GTumbush, 2020-09-04:
Fixed due to pull request https://github.com/openhwgroup/cv32e40p/pull/464 merged with master.

closed 2020-09-04 by GTumbush