openhwfoundation/cv32e40p #435: Which commit ID is stable for commerical use?

islengxw, 2020-08-03:
Hi openhwgroup,
Thanks for your hard working for this great project, i have learned GreenWaves have tapped out sucessfully using this RISCV core.
i am coming from chinese company, we are very instresting about this core for commerical usage.
Can anybody can give me some clue about which commit ID was stable enough for commerical use? BTW, must support floatting point.

MikeOpenHWGroup, 2020-08-06:
Hi @islengxw, the answer to your question depends on what you mean by "stable for commercial use".

The first release of the CV32E40P will not support floating point.   This is planned for a future release and OpenHW has not committed to date for this yet.

If you want to use the RTL as IP for an ASIC, then the CV32E40P is not yet ready.  We are working towards this goal and expect to be there in October, 2020.

If you want to integrate it into your own development environment, I would say it is more than stable enough for this.   Keep in mind that if you do, you will want to merge in updates before you build an ASIC with it.

If you want to port the device to an FPGA, I would wait a bit.  Our Hardware Task Group is currently porting the core to a Xilinx FPGA and you may as well let them find all the problems.  :wink:

If you want to experiment with the device in simulation, you'll find the CV32E40P to be very stable.   There are open bugs, but none are show-stoppers for experimentation-type work.

As we get closer to completion of this core (October), we will publish updates here and on the OpenHW Group website.

If this answers your question, kindly close the issue.   Thanks!

islengxw, 2020-08-07:
Hi Mike,
Very detailed and clear answer, i appreciate it very much. :)
BR
Xiaowei Leng

closed 2020-08-07 by islengxw