openhwfoundation/cv32e40p #439: Illegal Instruction Exception Raised Incorrectly - C.EBREAK

shetalani, 2020-08-06:
**RISC-V Specification:**

**Issue Description:**

Executing C.EBREAK instruction raises wrongly an illegal instruction exception. 

**Example:**

As shown below, the instruction 32'h00100073 (EBREAK) is decoded at time point t##0, where MSTATUS, MCAUSE and MEPC are updated at t##2 as if it had caused an illegal instruction exception.

[image]

**Top Level Parameters**
```
cv32e40p_core #(
    .FPU                      ( 0 ),
    .NUM_MHPMCOUNTERS         ( 1 ),
    .PULP_CLUSTER             ( 0 ),
    .PULP_XPULP               ( 0 ),
    .PULP_ZFINX               ( 0 )
) 
```
---
Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2020.1.4

Silabs-ArjanB, 2020-08-19:
Hi @shetalani , could you check if this issue has been fixed now on the not yet merged https://github.com/Silabs-ArjanB/cv32e40p/tree/ArjanB_illegal_instr_exc branch. If not, can you show a new violating waveform (and I will remove the resolved label again).

Silabs-ArjanB, 2020-09-02:
[image]

closed 2020-09-02 by Silabs-ArjanB