openhwfoundation/cv32e40p #444: Duplicate cv32e40p_sim_clock_gate.sv in compilation | Component:Tool-and-build

dawidzim, 2020-08-11:
Hey!
In `cv32e40p/example_tb/core/Makefile` is:
```
RTLSRC_BHV		:= $(wildcard $(RTLSRC_HOME)/bhv/*.sv)

RTLSRC			:= $(RTLSRC_HOME)/bhv/cv32e40p_sim_clock_gate.sv $(filter-out $(RTLSRC_HOME)/rtl/cv32e40p_register_file_latch.sv,\
				$(wildcard $(RTLSRC_HOME)/rtl/*.sv))
```
File `cv32e40p_sim_clock_gate.sv` already is in `RTLSRC_BHV`. I think this is not necessary duplicate this file.
So we can just delete this line `$(RTLSRC_HOME)/bhv/cv32e40p_sim_clock_gate.sv` or use `filter-out` in `RTLSRC_BHV`

Silabs-ArjanB, 2020-08-11:
Hi @dawidzim , thanks for spotting this. That is not correct indeed; I think it is cleanest to remove the $(RTLSRC_HOME)/bhv/cv32e40p_sim_clock_gate.sv part. Could you make a pull request for that?

dawidzim, 2020-08-11:
Issue will be solved by #446, after merge I will close this issue

Silabs-ArjanB, 2020-08-28:
@dawidzim The pull request has been merged now; could you please close this issue?

dawidzim, 2020-08-31:
Issue fixed

closed 2020-08-31 by dawidzim