openhwfoundation/cv32e40p #445: Stepie bit in DCSR not implemented

silabs-oysteink, 2020-08-11:
When single stepping, interrupts are taken regardless of the value written to the stepie bit in dcsr. 

By looking at the rtl, the stepie bit is defined, but never used.

CV32E40P hash: 916d92afc6bbc27b7ab65c503043982e1b6e3ab0

Silabs-ArjanB, 2020-09-04:
Hi @silabs-PaulZ @silabs-oysteink What was the conclusion again on this one; do we want stepie to be WARL (0), WARL (1) or will we support both values? (Reason I ask is that I am working on #481 and I can maybe fix this right away)

silabs-oysteink, 2020-09-04:
Hi,

The debug spec says: "Implementations may hard wire this bit to 0. In that case interrupt behavior can be emulated by the debugger". In our current state the rtl behaves as if it was tied to 1, not sure if that is allowed. I suggest we either tie it to zero, never take interrupts during single stepping, or support both values. First option would be less time consuming and less to verify.

Silabs-ArjanB, 2020-09-04:
Agreed with @silabs-oysteink ; my preference would be the same. I will tie the bit to 0 (so make it WARL (0)) as part of #481 ; @silabs-PaulZ can you then perform the actual RTL update to implement matching behavior (i.e. not allowing interrupts during single stepping)?

Silabs-ArjanB, 2020-09-04:
Partially fixed now in https://github.com/openhwgroup/cv32e40p/pull/483. Only interrupts need to be actually disallowed.

Silabs-ArjanB, 2020-09-09:
Fixed by fully implementing the stepie functionality (so it is R/W now). Related pull request https://github.com/openhwgroup/cv32e40p/pull/494 has been merged.

Silabs-ArjanB, 2020-09-24:
Hi @silabs-oysteink Can you please close this issue if you agree it has been fixed?

closed 2020-09-24 by silabs-oysteink