openhwfoundation/cv32e40p #447: What is the basis for a TAG?

islengxw, 2020-08-13:
Hi OpenHWGroup members,
I understand we should tag some commit ID for released version, which should be stable or after the regression test.
But in this project, i just wondering how many tests(for example SIMULATION, FPGA, synthesis, even PR?) should pass before we can make a tag for a release version?
 
Thanks inadvance~~
BR
Xiaowei Leng

Silabs-ArjanB, 2020-08-13:
Hi @MikeOpenHWGroup , can you answer this one please?

MikeOpenHWGroup, 2020-08-13:
Thanks for the question @islengxw.  There are many different strategies for creating tags and the CV32E40P development team has not yet defined how we should tag our code.  But I think it is time to do so, so I will add some thoughts here based on what I have done at the past at other organizations.

My proposal is that we should tag the code base every day with a date.  For example, cv32e40p_20200813.  There does not need to be any requirements defined for creating the tag.  Ideally, the tag would be created automatically by a script (does GitHub support this?).

This strategy is simple and gives the user of a tag the ability to control the code they use and track its history.

Here are a few examples of how this tagging strategy can work:

1. Your team may wish to create an FPGA prototype of CV32E40P.   Your criteria for this may be as simple as running the "hello-world" test and a lint-check for synthesis.  So you clone the latest tag, run "hello-world" and "lint" and if these pass, you can proceed.
2. After some days in the lab with your FPGA, you find an RTL bug and you report it as an issue to the CV32E40P project on GitHub.  Providing the tag  of the code you used helps to debug.  Also, the tagging system makes it easy to see when your issue was resolved and to know which tags have the bug and which tags have the fix.
3. As some point you may wish to tape-out an ASIC based on the CV32E40P RTL.  In this case you will want to use a tag with a known history of complete verification (both for functionality and physical design).  Because both the RTL and verification IP are open source, you can either do this yourself, or you can work with OpenHW to get it done.

I am interested you hear your thoughts on this.

bluewww, 2020-08-13:
A commit (hash) already uniquely defines a point in the repository's history. You can easily look up at what poin in time the commit was created. If you just tag the latest commit daily there won't be much use over just using/referring to commit hashes directly.

I believe that tags should be reserved to ascribe certain qualities to a commit hash (or point in time, if you will). Release tags are a useful concept in that case, which I belive is related to the question of the original poster (@islengxw). People using release tags can be sure that certain kinds of tests have been performed on this version (all tests passing, synthesis, trial backend etc.). What kind of assurance should be provided is to be discussed.

islengxw, 2020-08-20:
Hi Mike,
Thanks for you reply. I understand the strategy now.
I agree with @bluewww , i am also considering there won't be much use for daily tags since we can refer to the commit hashes directly.

Silabs-ArjanB, 2020-09-24:
Hi @islengxw We will not do daily tags. We will tag once the (currently being discussed) functional verification and RTL design criteria have been met (e.g. basic synthesis check, tests implemented as per verification plan and passing, etc.). We will document the status when we apply such a tag. 

Can you please close this ticket if your question has been answered?

islengxw, 2020-09-24:
Hi ArjanB,
Thanks

closed 2020-09-24 by islengxw