openhwfoundation/cv32e40p #448: mcounteren CSR should not exist

Silabs-ArjanB, 2020-08-13:
The official RISC-V spec was updated in response to https://github.com/riscv/riscv-isa-manual/issues/561, see https://github.com/riscv/riscv-isa-manual/commit/1eaebccc78d88f110a68faf1b39938fc291a4065 and comments on that commit).

We should remove mcounteren from CV32E40P (this issue relates to #432 as well, but the related questions have not been answered yet in the RISC-V organization).

MikeOpenHWGroup, 2020-08-13:
I would be happy about this.  `mcounteren` is not currently modeled in the Reference Model and the CSR does nothing in the core (other than return 0).   Would removing this CSR mean that reads to 0x306 would invoke an illegal instruction exception?

Silabs-ArjanB, 2020-08-13:
Yes that would be the consequence if we change this

MikeOpenHWGroup, 2020-08-13:
Then I am in favor of this.

closed 2020-08-31 by Silabs-ArjanB