openhwfoundation/cv32e40p #452: Some of the Xpulp instructions use opcodes that are either reserved opcodes or standard opcodes

Silabs-ArjanB, 2020-08-19:
Some of the Xpulp opcodes have been chosen such that upstreaming of the related tool changes will become impossible unless we change some opcodes. Ideally only 'custom opcodes' are used for Xpulp extensions.

First question to answer:

- [ ] Are there enough custom opcodes  to host all the Xpulp extensions that are now spread in the reserved and standard opcodes?

Craig Blackmore's original message:

As per the actions of the minutes above, here is a summary of opcode issues followed by a breakdown of opcodes used by each CORE-V extension.

The issues fall into 4 categories:

1. Use of standard opcode (e.g. OP, LOAD, BRANCH). Any CORE-V extension that uses these opcodes may conflict with a future RISC-V standard extension.
2. Use of reserved opcode. Any CORE-V extension that uses a reserved opcode may conflict with a future RISC-V standard extension.
3. Use of custom-<n> opcode. The RISC-V Foundation does not allow any standard extension to use the custom opcodes. Therefore, any CORE-V extension that only uses custom opcodes is guaranteed not to conflict with future standard extensions. However, any custom opcodes must be changed before a CORE-V extension can be accepted as a standard extension.
4. Incorrect use of standard opcode. The bitmanip immediate instructions (p.extract, p.extractu, p.insert, p.bclr, p.bset) use OP, but they should use OP-IMM (if we choose to continue using standard opcodes).

Any CORE-V extension that uses standard or reserved opcodes risks conflicting with future standard extensions. Using custom opcodes eliminates that risk, but those opcodes must be changed if the extension is to be standardized. I suspect we would also need a compelling reason to use a reserved opcode rather than fitting into the existing ones. Crucially, whatever encodings we choose now are subject to change during standardization.

Here is a summary of opcodes used by each CORE-V extension:

- PULP bitmanip
   All instructions use OP.
   Immediate instructions (p.extract, p.extractu, p.insert, p.bclr, p.bset) should use OP-IMM.

- PULP SIMD - reserved opcode.

- Post-increment load/store - custom-0 opcode.

- PULP general ALU - Add-shift instructions (p.add[u][R]N[r] and p.sub[u][R]N[r]) use custom-3 opcode.
   Multiply-accumulate instructions use custom-2 opcode.
   Remaining instructions use OP opcode.

- PULP immediate branching - BRANCH opcode.

- PULP HW loop - custom-3 opcode.

- Xpulpcluster - LOAD opcode.

jm4rtin, 2020-08-19:
I haven't looked too close at how the branch is encoded but if you exclude it you *might* be able to squeeze it. It might be a tight fit and may make some encoding choices a bit odd due to tightly packing the space. If you add the branch back I'm not sure. There are a few instructions that are grouped together where some have an immediate and others have this immediate tied to 0. This is wasteful and they can be moved to a different encoding type but will no longer be logically grouped together. We may need to try remapping them and proposing something to see if some instructions need to be cut.

Spreadsheet I used for my rough estimation: [pulp_encoding_blocks.xlsx](https://github.com/openhwgroup/cv32e40p/files/5099568/pulp_encoding_blocks.xlsx)

davideschiavone, 2020-08-20:
Thanks a lot @Silabs-ArjanB and @jm4rtin ! Very useful thread.
In my understanding, we should try to see whether we can squeeze all the RESERVED and STANDARD instructions into the custom space.

It's a big task :) However, if there is not enough space, we can think of getting rid of some "less-useful" instructions, and/or reducing the immediate where possible.
I am not saying it makes sense, but for sure we know that at this stage they won't get accepted to the standard GCC.

As this is a long term task (I don't think we will remap all of them over the next month), we need to be conscius that products and prototypes being developed now must rely on the PULP GCC, that is not maintained by OpenHW.
In addition, as we will modify the CV32E40P spec, this also means that OpenHW won't maintain the current ISA of the CV32E40P (with the current encoding etc)

Just a note

jm4rtin, 2020-08-20:
It should be possible to remap them all. Here is my first attempt at remapping them: [pulp_encoding_blocks.xlsx](https://github.com/openhwgroup/cv32e40p/files/5105416/pulp_encoding_blocks.xlsx)

Some notes:
* Decoding would start with the block (`custom-0` ... `custom-3`)
* Next would be the `funct3` field
* After that it depends on the block/plane:
  * For `custom-0` it maps to an instruction
  * For `custom-1` it either maps to an instruction or goes to `Plane A` or `Plane B`
  * For `custom-2` it looks at `f3` to determine the instruction
  * For `custom-3` it uses a combination of `funct5` and `F` to determine the instruction (this is the SIMD instructions moved directly to `custom-3`)
  * For `Plane A` it looks at `funct7` to determine the instruction
  * For `Plane B` it looks at `funct4` to determine the instruction (this plane is only used by the hardware loop right now)
* If you exclude SIMD a vendor could re-purpose `custom-3` for their own use
* A grouping of instructions (e.g. bit manipulation) may no longer reside in a single block
* A couple encodings for instructions have moved the bits around slightly
* This may add a bit of complexity (read: longer combinatorial path) to a few instructions
* Some of the instructions may need to be moved around in the encoding space a bit to simplify the mapping of bits in the instruction to control signals for the data path

Thoughts?

Silabs-ArjanB, 2020-08-21:
Hi @jm4rtin Thank you for this great proposal. Of course it is difficult to foresee if timing issues would arise from this; it would probably be easiest to just try it out to figure that out. I like that SIMD is only in custom-3 and that no other instructions are in that group. It would be nice if such a property was true for bit manipulation as well (as we know that a standard for that will be ratified), but I do not see that as a requirement. You could even argue that we should not focus on SIMD and bit manipulation too much and use those instructions to 'fill the holes' and that the focus should be on having a clean encoding for all the other instructions.

Silabs-ArjanB, 2020-09-02:
Once this issue has been resolved the assertions from #457 can be checked again to see if they pass.

Suxiaojie-iot, 2021-01-21:
Recently, when I transplanted cv32e40p on the cluster side of Pulp, I found some problems. In addition to the interface mismatch, some internal logic is also different, including the fetching and decoding stages.
In this regard, I would like to ask how to implement Pulp's cluster kernel to cv32e40p.
Thank you.

MikeOpenHWGroup, 2021-01-21:
Hi @Suxiaojie-iot, the items you mention are deliberate changes, not problems.  Please refer to the [cv32e40p user manual](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/index.html).

Also, in the future, please do not mix multiple topics in a single GitHub issue.   This issue is for Xpulp instruction opcodes.  To discuss integration of "Pulp's cluster kernel to cv32e40p" you should open an new issue.

Suxiaojie-iot, 2021-01-22:
Letter received. Sorry for doing so. This is the first time, I will pay attention next time, thank you.

davideschiavone, 2022-03-08:
@jm4rtin , @jeremybennett any comment on this?

pascalgouedo, 2022-04-06:
Hi,
Here is a new version [pulp_encoding_blocks-OPHW-jm-jpb-pgo-2022-04-06.xlsx](https://github.com/openhwgroup/cv32e40p/files/8428465/pulp_encoding_blocks-OPHW-jm-jpb-pgo-2022-04-06.xlsx) with X's removal and some bit manip instructions grouping in custom-3.

pascalgouedo, 2022-04-26:
Last update on pack instructions [pulp_encoding_blocks-OPHW-jm-jpb-pgo-2022-04-26.xlsx](https://github.com/openhwgroup/cv32e40p/files/8565176/pulp_encoding_blocks-OPHW-jm-jpb-pgo-2022-04-26.xlsx)

pascalgouedo, 2022-04-27:
Thanks John.
#700 will only be documentation updates.
I will submit another PR with RTL changes.

pascalgouedo, 2022-04-28:
#700 replaced by #704.

pascalgouedo, 2022-09-30:
Hi
As always when starting to implement specification and verifying it, you find some problems you didn't anticipate.
Here things are that some ALU control signals are not generated in decoder file but in id one and were not analysed/taken into account in previous versions of re-encoding tables (at least by me).

Here is a new [pulp_encoding_blocks-OPHW-2022-09-26.xlsx](https://github.com/openhwgroup/cv32e40p/files/9683505/pulp_encoding_blocks-OPHW-2022-09-26.xlsx) version taking into account ALU control signals.
First tab lists modifications with respect to older versions and each tab has a comment for each changed instructions.

Best regards,
Pascal.

pascalgouedo, 2022-11-07:
Hi,
The very last version [pulp_encoding_blocks-OPHW-2022-10-07.xlsx](https://github.com/openhwgroup/cv32e40p/files/9949531/pulp_encoding_blocks-OPHW-2022-10-07.xlsx) with final encoding of the 2 new Hardware Loops instructions.

davideschiavone, 2023-08-04:
@pascalgouedo can we close this issue?

pascalgouedo, 2023-08-23:
Resolved in cv32e40p_v1.2.0

closed 2023-08-23 by pascalgouedo