openhwfoundation/cv32e40p #454: How to handle the error response with cv32e40p?

islengxw, 2020-08-20:
Hi,
I have noticed that there are instr_err_i and data_err_i ports in the IBEX(another RISCV core developped under lowRISC), should cv32e40p also support this?
For example, the core communicate with peripherals on the AMBA system bus, the slave can feedback the error response signal, how the cv32e40p handle this?

Thanks
Xiaowei Leng

Silabs-ArjanB, 2020-08-20:
Hi @islengxw This is a feature that we would like to have indeed. It has been postponed to a next core though, see for example https://github.com/openhwgroup/cv32e40p/issues/343. The way to deal with such bus errors currently is to convert them into interrupts (by which of course the errors will no longer be precise). If you want to start implementing this feature for a next core, e.g. CV32E40, please let us know. Could you please close this ticket as it is basically a duplicate?

Silabs-ArjanB, 2020-08-31:
Closing this duplicate ticket

closed 2020-08-31 by Silabs-ArjanB