openhwfoundation/cv32e40p #455: Bits 31:0 of mhpmcounter3 are not 0

darbaria, 2020-08-25:
Bits 31:0 of mhpmcounter3 are not 0, violating the requirement specified in Section Machine Performance Monitoring Counter
All other counters satisfy this requirement. Branch tested 16 July cv32e40p. 

[mhpm_counter3_lsb32_bits_issue.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5123917/mhpm_counter3_lsb32_bits_issue.vcd.gz)

### Component
CSR registers

### Steps to Reproduce
Used formal property checking
[mhpm_counter3_lsb32_bits_issue.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5123921/mhpm_counter3_lsb32_bits_issue.vcd.gz)

Silabs-ArjanB, 2020-08-25:
Hi @darbaria Why do you expect that mhpmcounter3  is 0? In the trace you provided it seems to me that there is a regular CSR write access that writes the mentioned counter to a non-zero value. I can't see how the parameters to the CV32E40P are set, but using the default settings would exclude the counters 4 upwards which would explain why those counters might seem different. If you still think there is an issue, can you please be more explicit why you think it is not okay for mhpmcounter3 to be non-0 in the provided trace?

darbaria, 2020-08-25:
Hi Arjan

I'm looking at the spec (attached snapshot) which says that 31:0 is 0 for counters ranging from mhpmcounter3 to mhpmcounter31. Other counters do have this property, as I can see the proofs.

Ashish

jm4rtin, 2020-08-25:
The attached snapshot didn't come through. It would be better to reply on the GitHub website. Do you mean this description in the CSR section that simply states "0"?

[image]

That probably needs to get updated. If the performance counter is implemented (it is by default) the register is fully read/write and will also increment if not inhibited by `mcounterinhibit`. Which events cause it to increment are determined by `mhpmevent3`. See the chapter on the HPM for a better description of how it works: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/perf_counters.html

darbaria, 2020-08-25:
Yes, @jm4rtin I was referring to that.

Silabs-ArjanB, 2020-08-25:
Hi @darbaria Can you let us know the setting of   parameter NUM_MHPMCOUNTERS? If it is set to 1 then mhpmcounter3 is implemented and can be written to a non-zero value. I agree that above description of '0' does not make sense. Are you okay if we change this issue into a 'documentation only issue' or do you think there is still an actual RTL issue?

darbaria, 2020-08-25:
Hi @Silabs-ArjanB yes, it is set to 1. Can you please revise the spec on https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html#cs-registers? Yeah, you may move this to a doc issue as this is what it appears to be.

Silabs-ArjanB, 2020-08-26:
Hi @darbaria Yes, we will fix the documentation. Thank you for raising this issue. I will move this github issue to the core-v-docs repos.